tinygo-org/drivers #14: Need a way to override the default I2C address

deadprogram, 2019-01-29:
There are several cases where an i2c device can have an alternative i2c address, due to manufacturer settings or variation, or my user jumper settings.

Perhaps this could be changing:

```go
type Device struct {
	bus machine.I2C
        Address uint16
}

func New(bus machine.I2C) Device {
	return Device{bus: bus, Address: Address}
}
```

Which would allow user to do something like this:

```go
sensor := mag3110.New(machine.I2C0)
sensor.Address = 0xC2 // some alternative if needed
sensor.Configure()
```

We have to change the usage of `Address` in the methods to `d.Address` as well.

aykevl, 2019-01-29:
This sounds like a good idea!
Of course, this depends on device support but as far as I know most devices have some way to adjust the I2C address.

deadprogram, 2019-03-11:
Also, sometime "clones" use a different address. I recall that being the case with some MPU6050s and BMP180's that I have gotten from Alibaba.

deadprogram, 2019-04-08:
This is now in the `master` branch. Closing.

closed 2019-04-08 by deadprogram