tinygo-org/drivers #18: Tests

deadprogram, 2019-02-07:
We need to have a way to run tests against these drivers...

deadprogram, 2019-03-11:
If we were to run "smoke" tests by building the example for each driver, what would be the best way to do this? Download the latest released pre-built binary version of TinyGo from https://github.com/tinygo-org/tinygo/releases or build a dev version from source, or ?

aykevl, 2019-03-11:
I think downloading the release is both faster and more reliable.

deadprogram, 2019-04-22:
I have implemented the build of one example per driver as a smoke test. Build is located here: https://circleci.com/gh/tinygo-org/drivers

I will leave this issue open, as we still need more automated tests, this is just a starting point.

closed 2022-08-16 by deadprogram