tinygo-org/drivers #27: Display interface discussion

conejoninja, 2019-03-09:
This is not an issue, but a discussion on how the _Display_ interface, common interface for managing screens, should look. I really like to heard opinions

I've been looking at [Adafruit's GFX library](https://github.com/adafruit/Adafruit-GFX-Library), the only required function for a display driver is _drawPixel(x,y,color)_ , everything else could be done by GFX itself or overridden by the driver. There are two sets of functions _drawXXX_ and _writeXXX_ but I'm not sure what is the difference between them (I guess one writes in the buffer and the other send it to the screen), or what _startWrite_ and _endWrite_ do.  Also, the 3 drivers I saw (2dom/PxMatrix, adafruit/PCD8544, or adafruit/SSD1306) all use the _display_ method (which send the buffer to the screen), but it's not in GFX specification.

_Display_ interface should be as small as possible, and while related, should not be mistaken by a GFX-like library.

This is part of the Adafruit_GFX.h
```C
// This MUST be defined by the subclass:
  virtual void drawPixel(int16_t x, int16_t y, uint16_t color) = 0;

  // TRANSACTION API / CORE DRAW API
  // These MAY be overridden by the subclass to provide device-specific
  // optimized code.  Otherwise 'generic' versions are used.
  virtual void startWrite(void);
  virtual void writePixel(int16_t x, int16_t y, uint16_t color);
  virtual void writeFillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color);
  virtual void writeFastVLine(int16_t x, int16_t y, int16_t h, uint16_t color);
  virtual void writeFastHLine(int16_t x, int16_t y, int16_t w, uint16_t color);
  virtual void writeLine(int16_t x0, int16_t y0, int16_t x1, int16_t y1, uint16_t color);
  virtual void endWrite(void);

  // CONTROL API
  // These MAY be overridden by the subclass to provide device-specific
  // optimized code.  Otherwise 'generic' versions are used.
  virtual void setRotation(uint8_t r);
  virtual void invertDisplay(boolean i);

  // BASIC DRAW API
  // These MAY be overridden by the subclass to provide device-specific
  // optimized code.  Otherwise 'generic' versions are used.
  virtual void
    // It's good to implement those, even if using transaction API
    drawFastVLine(int16_t x, int16_t y, int16_t h, uint16_t color),
    drawFastHLine(int16_t x, int16_t y, int16_t w, uint16_t color),
    fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color),
    fillScreen(uint16_t color),
    // Optional and probably not necessary to change
    drawLine(int16_t x0, int16_t y0, int16_t x1, int16_t y1, uint16_t color),
    drawRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color);

```

Therefore, I propose the following interface
```go 
type Display interface {
	// show the pixel on the screen
	DrawPixel(x uint16, y uint16, color uint16) 

	// modify the buffer, not the screen
	WritePixel(x uint16, y uint16, color uint16)
	WriteLine(x0 uint16, y0 uint16, x1 uint16, y1 uint16, color uint16)
	WriteFastVLine(x uint16, y uint16, length uint16, color uint16)
	WriteFastHLine(x uint16, y uint16, length uint16, color uint16)
	WriteFillRect(x uint16, y uint16, w uint16, h uint16, color uint16)
	Display() // send the buffer (if any) to the screen (instead of startWrite/endWrite)
	DisplayFor(ms uint16) // send the buffer (if any) to the screen for _ms_ milliseconds (some devices may need a time), not sure if this make sense or how to merge with _Display()_
}
```
These other functions could go on a gfx-like package and be overridden by the device driver if needed
```go 
type GFX interface {
	// These second group is more abstract, and while it can be overridden, it's usually left on GFX
	DrawFastVLine(x uint16, y uint16, length uint16, color uint16)
	DrawFastHLine(x uint16, y uint16, length uint16, color uint16)
	FillRect(x uint16, y uint16, w uint16, h uint16, color uint16)
	FillScreen(color uint16)
	DrawLine(x0 uint16, y0 uint16, x1 uint16, y1 uint16, color uint16)
	DrawRect(x uint16, y uint16, w uint16, h uint16, color uint16)
	SetRotation(rotation uint8)
	InvertDisplay(invert bool)

	// These are pure-GFX work, not in the drivers
	DrawCircle(x uint16, y uint16, r uint16, color uint16)
	FillCircle(x uint16, y uint16, r uint16, color uint16)
	DrawRoundRect(x uint16, y uint16, w uint16, h uint16, r uint16, color uint16)
	FillRoundRect(x uint16, y uint16, w uint16, h uint16, r uint16, color uint16)
	DrawTriangle(x0 uint16, y0 uint16, x1 uint16, y1 uint16, x2 uint16, y2 uint16, color uint16)
	FillTriangle(x0 uint16, y0 uint16, x1 uint16, y1 uint16, x2 uint16, y2 uint16, color uint16)

	 // These are probably better suited for a TextDisplay interface
	DrawChar(x uint16, y uint16, c byte, color uint16, bg uint16, size uint8)
	SetCursor(x uint16, y uint16)
	SetTextColor(color uint16)
	SetTextBackgroundColor(color uint16)
	SetTextSize(size uint8)
	SetTextWrap(w bool)
	SetFont()

	DrawBitmap(x uint16, y uint16, bitmap []uint8, w uint16, h uint16, color uint16)
}
```

deadprogram, 2019-03-09:
2 frameworks I have noticed recently:

https://github.com/littlevgl/lvgl
https://github.com/idea4good/GuiLite

Just to throw that into the mix...

aykevl, 2019-03-09:
That's a good starting point! I'm sorry I've ignored the display driver for so long, it's important to have. But I also care a lot about a clean API.
Long wall of text following...

My first idea was to keep the buffer inside the graphics library, but on second thought maybe it's better after all to keep it in the driver itself because of all the different pixel formats. A driver could expose generic low-level primitives that can either be used directly or be used by a graphics library. I certainly would want to avoid double-buffering, as such buffers are very big. Let me know what you think about the pros/cons of both.

Some thoughts about the coordinate integer type:

* While a 16-bit integer is probably [big enough for all practical purposes](https://www.quora.com/What-is-the-highest-resolution-and-biggest-size-desktop-monitor-today), a 32-bit integer might be slightly faster on 32-bit microcontrollers (but significantly slower on 8-bit microcontrollers). I don't think we have to care about 16-bit microcontrollers here, they aren't used much nowadays. The reason 32-bit integers aren't an overhead on 32-bit systems is because in general RISC systems (almost all MCUs) don't support anything but 32-bit integer operations and those are usually performed in a single cycle.
* Using `int` would make the interface slightly more intuitive to work with (`int` is equivalent to `int32` in TinyGo on <=32bit systems).
* Using signed integers may make some calculations easier to perform, for example for negative offsets from a given location. Even with 16-bit integers, I think a range of ~32767 is big enough for all displays.

The color type you used there is also a `uint16`, which I think is too limiting: it would limit the interface to displays with only a certain color format, and color formats vary widely between displays. Instead, I would opt for either a `uint32` in RGB format (`0xffaa00` for orange, for example) or re-use something like `color.RGBA`.

Regarding the API names, to me a `Write` prefix means actually writing something to the screen (think of `io.Writer`). Instead, I would suggest using a `Set` or `Draw` prefix, for example `SetPixel`.

Regarding the API itself, I think we should keep it as small as reasonably possible. The Adafruit GFX library is written in C++ so can use inheritance which means you can actually write generic functions that get overridden in specific drivers. Go doesn't have such things. However, we can define generic helper functions that do these operations and do a type assert on the display driver. As long as only one display diver is in use in a program, TinyGo should optimize all this abstraction away to direct calls and should even perform inlining where profitable.

So here is my counter-proposal:

```go
import "display/color"

type Display interface {
	// Get the current size of the display.
	// Perhaps we also want an optional function to get DPI if known?
	Size() (x, y int16)

	// Modify the internal buffer.
	SetPixel(x, y int16, c color.RGBA)

	// Send the buffer (if any) to the screen (instead of startWrite/endWrite).
	Display() error
	// Send a single pixel to the screen.
	DisplayPixel(x, y int16, c color.RGBA)
	// Send this part of the buffer to the screen for incremental updates
	// (e.g. button press animations, blinking cursor, etc.).
	DisplayRect(x1, y1, x2, y2 int16)
}

// SetLine sets a line in the display to the given color.
// (Note: there should probably be two implementations of this, of which one
// does anti-aliasing).
func SetLine(d Display, x1, y1, x2, y2 int16, c color.RGBA) {
	if x1 > x2 {
		x1, x2 = x2, x1
	}
	if y1 > y2 {
		y1, y2 = y2, y1
	}
	if x1 == x2 {
		// fast path for vertical line
		for y := y1; y <= y2; y++ {
			d.SetPixel(x1, y, c)
		}
	} else if y1 == y2 {
		// fast path for horizontal line
	} else {
		// slow path with 
	}
}
```

> 2 frameworks I have noticed recently:
> 
> https://github.com/littlevgl/lvgl
> https://github.com/idea4good/GuiLite
> 
> Just to throw that into the mix...

Those are really cool, especially LittlevGL looks really attractive. We should have something like that in Go, or write good wrappers :smile:

conejoninja, 2019-03-12:
I like @aykevl proposal, it's simple and a good starting point, will start working on it

conejoninja, 2019-04-10:
On hold until https://github.com/tinygo-org/tinygo/issues/270 is fixed

conejoninja, 2019-04-21:
Looks like https://github.com/tinygo-org/tinygo/issues/270 is not easy to fix (and I do not know the reason, as copying the content of geom.go into any other package seems to work). Looking at https://golang.org/pkg/image/ Point is mostly used in geometric operations, (X,Y) coordinates are use in pixel manipultion (AlphaAt(x,y), PixOffset(x,y), Set(x,y, color), SetAlpha(x,y,alpha),...) If it was fixed, I would propose to use image.Point only as result of Size() (p Point) {}

To move forward with the discussion, and be able to accept / submit some display drivers, I propose the following interface

```go
type Display interface {
	// Size returns the current size of the display.
	Size() (x, y int16)

	// SetPizel modifies the internal buffer.
	SetPixel(x, y int16, c color.RGBA)

	// Display sends the buffer (if any) to the screen.
	Display() error

	// DisplayPixel sends a single pixel to the screen.
	DisplayPixel(x, y int16, c color.RGBA)
}
```

We drop image.Point in favor of X,Y int16. 
(changes are needed in #31 -drop DisplayRect-, #32 is compatible and could be accepted)

deadprogram, 2019-04-22:
Let's rename the interface from `type Display interface` to be `type Displayer interface` to more consistent with Go interface naming, and then use the interface you propose here, @conejoninja like you said in the interest of moving forward.

conejoninja, 2019-04-22:
Changes done in #31

deadprogram, 2019-04-24:
The simplified interface is now in the `master` branch. Closing this issue.

closed 2019-04-24 by deadprogram