tinygo-org/drivers #37: BME280 implementation

Munsio, 2019-04-18:
Hi, 

I'm currently trying to implement a bme280 driver.
This is the first time working on something like that and i have a few problems which i like to discuss on.

I started the whole process by "copying" the bmp180 code and tinker around with the registers i found at: https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/bmp280_driver.go

And here is the Datasheet:
https://www.embeddedadventures.com/datasheets/BME280.pdf

There are 2 things i found:
1) every time I'm using the `sensor.Connected` function the whole program stops working. Reading out the temperature in an for loop works without problems

2) i copied the configuration set and temperature calculation code from the gobot project but the resulting value is 10° to low.

I will link an WIP Pull Request to this issue

aykevl, 2019-04-18:
On which chip are you running this?
Most ARM Cortex-M chips can be used with GDB, which allows single stepping through the program or breaking it to find where it hangs.

Munsio, 2019-04-18:
I use an Arduino Uno - currently i am building the hex file with the docker container cause of my panic during installation. As soon as i can fix that i will try it with a debugger - thanks

aykevl, 2019-04-18:
I see. Unfortunately, the AVR backend has many bugs and I'm not aware of usable AVR debuggers (without expensive hardware). These bugs may result in taking the wrong side of an if/else, for example. I think you're better off using an ARM based chip (any Cortex-M, like the ones branded "M0" or "Zero"): hopefully the AVR backend will mature to be usable but we're not there yet.

ssorc, 2019-05-29:
> I started the whole process by "copying" the bmp180 code

There is a BME280 driver in the gobot repository that may be a better reference than the BME180 driver that you have used:

https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/bme280_driver.go

closed 2019-06-04 by Munsio