tinygo-org/drivers #51: RGBW leds

directionless, 2019-05-07:
I was playing with rgbw leds, using the existing `ws2812` driver. It turned out to be quite easy, instead of writing the GRB bytes,  I needed to write a 4th W byte. So, what might support for this look like?

I imagine there's some benefit to using the existing `color.RGBA` type. Does adding an optional parameter for the white array make sense?

deadprogram, 2019-05-07:
I just wonder if in this case, we should use the alpha channel for that? Your typical WS2812 does not support which is why `WriteColors()` currently ignores that data. Seems like the easiest thing to do, probably, would be to add an additional member to the struct type like `HasWhite bool` which would control if 3 or 4 bytes are sent.

What do you think?

aykevl, 2019-05-07:
Perhaps even better: an enum to describe the color model (or however this should be called):

- GRB (currently hardcoded)
- GRBW (your typical RGBW LED strip)

Some (older) strips do not follow the GRB order and it would be useful to add support for those without breaking the API.

deadprogram, 2019-05-07:
That's a much better idea.

aykevl, 2019-05-07:
Still, I wonder how we can best pass the white channel in this API. We could (ab)use the alpha channel for that, but I think that's wrong. Maybe we should define a new color type for that, for RGBW?

Come to think of it, I suspect the APA102 uses the wrong color format. I think it should use NRGBA instead of RGBA because it essentially does the alpha calculation inside the LED.

syoder, 2020-01-16:
I have a 5x8 Neopixel RGBW panel that I was trying to get working and ran into this issue. And my initial thought was maybe it's not completely crazy to use the alpha channel for white.  But if we want to avoid that, we could keep it simple and do something like this:
```go
type RGBW struct {
	color.RGBA
	W uint8
}

func (d Device) WriteColorsRGBW(buf []RGBW) error {
	for _, color := range buf {
		d.WriteByte(color.G) // green
		d.WriteByte(color.R) // red
		d.WriteByte(color.B) // blue
		d.WriteByte(color.W) // white
	}
	return nil
}
```

That would leave the existing `WriteColors` as-is so that we wouldn't break it.

Thoughts? I'm happy to throw together a PR if there is some consensus on how to support RGBW.