tinygo-org/drivers #52: RGB Leds and Brightness

directionless, 2019-05-07:
Modern RGB leds are amazingly bright. This often leads to wanting to set some global brightness level. I'm not sure if this is more desirable as something using the alpha channel, or as something that could be set for all leds. 

I guess it's a bit lossy, since it would result in some floating point approximation. Still though, it seems valuable to me

deadprogram, 2019-05-07:
This could be done with a Luma calculation https://stackoverflow.com/questions/596216/formula-to-determine-brightness-of-rgb-color but not sure that this would go in any particular driver. Seems to me like this would be some application specific code.

aykevl, 2019-05-07:
Yes, I'm thinking the same. If the hardware has support for dimming we should use that in the driver, but otherwise leave it up to the application.
Also see this package that I've started to write, inspired by FastLED:
https://godoc.org/github.com/aykevl/ledsgo