tinygo-org/drivers #88: microbitmatrix: timing issues?

DazWilkin, 2019-09-19:
microbitmatrix is a good addition, thank you!

I'm struggling to use it as a counter and the issue appears to be timing-related.

The following mashes up the repo's [example](https://github.com/tinygo-org/drivers/blob/master/examples/microbitmatrix/main.go) and (at this incorrect speed) it works :

```Golang
package main

import (
	"image/color"
	"time"

	"tinygo.org/x/drivers/microbitmatrix"
)

var (
	display microbitmatrix.Device
	c = color.RGBA{255, 255, 255, 255}
)

func toMatrix(n int) {
	led := int16(0)
	for n > 0 {
		if n&1 == 1 {
			display.SetPixel(led/5, led%5, c)
		}
		n = n >> 1
		led++
	}
}
func main() {
	display = microbitmatrix.New()
	display.Configure(microbitmatrix.Config{
		Rotation: 0,
	})

	then := time.Now()
	for t := 0; ; t++ {
		if time.Since(then).Nanoseconds() > 80000000 {
			then = time.Now()
			display.ClearDisplay()
			toMatrix(t)
		}
		display.Display()
	}
}
```
However, I'd prefer something closer to per-second updates. But, any changes I make to the `time.Since(then)` condition result in seemingly random behavior from the matrix.

You should be able to observe this with:
```golang
time.Since(then).Seconds() > 1
```
is this explicable?

I'm able an mbed variant of this code without issue.

deadprogram, 2019-10-14:
Basically, in the current version of TinyGo the `time.Sleep()` method is what increments the time counter, so it is unfortunate but not surprising that the code in the provided example does not work as expected.

We've been discussing this at some length in the Slack channel as well in a couple of related GH issues.

I 

```go
package main

import (
	"image/color"
	"time"

	"tinygo.org/x/drivers/microbitmatrix"
)

var (
	display microbitmatrix.Device
	c       = color.RGBA{255, 255, 255, 255}
)

func toMatrix(n int) {
	led := int16(0)
	for n > 0 {
		if n&1 == 1 {
			display.SetPixel(led/5, led%5, c)
		}
		n = n >> 1
		led++
	}
}
func main() {
	display = microbitmatrix.New()
	display.Configure(microbitmatrix.Config{
		Rotation: 0,
	})

	then := time.Now()
	for t := 0; ; t++ {
		if time.Since(then).Seconds() > 1 {
			then = time.Now()
			display.ClearDisplay()
			toMatrix(t)
		}
		display.Display()
		time.Sleep(100 * time.Millisecond) // sleep here. will update the timer
	}
}
```

Not perfect solution, but at least for now does give the expected behavior.

conejoninja, 2019-10-14:
I've been a bit busy and didn't see the slack discussion about time.Sleep, should we update the example then? or it's something we're working on?

deadprogram, 2019-10-14:
That seems like a good idea...

conejoninja, 2019-10-14:
After some review, this issue doesn't happen on the microbitmatrix example. But I have some questions in this case, as *t* is being incremented several times per second and I'm not sure that's the expected behavior:

In this loop: 
```go
	for t := 0; ; t++ {
		if time.Since(then).Seconds() > 1 {
			then = time.Now()
			display.ClearDisplay()
			toMatrix(t)
		}
		display.Display()
	}
```
*t* is incrementing at a stable rate of 159 per second (microbit runs at 16MHz) each time we call toMatrix(t) 

If we add the time.Sleep(...)
```go
	for t := 0; ; t++ {
		if time.Since(then).Seconds() > 1 {
			then = time.Now()
			display.ClearDisplay()
			toMatrix(t)
		}
		display.Display()
		time.Sleep(100 * time.Millisecond) // sleep here. will update the timer
	}
```

Then it increases in steps of 10

In both examples, the increment of *t* depends on time.Sleep/code executed inside the loop (and processor speed). I'm thinking that maybe @DazWilkin wanted to show a new led each second, something like the code below (no time.Sleep needed, update t each second instead of multiple times per second):

```go 
	t := 0
	for  {
		if time.Since(then).Seconds() > 1 {
			then = time.Now()
			display.ClearDisplay()
			toMatrix(t)
			t++
		}
		display.Display()
	}
```

DazWilkin, 2019-10-25:
I'm embarrassed that I submitted that code; I don't know why I thought it would work.

@conejoninja is absolutely correct. That is what I intended and works as expected.

My apologies :-(

closed 2019-10-25 by DazWilkin