tinygo-org/drivers #100: compilation error when using tinygo.org/x/drivers/espat/mqtt

titimoby, 2019-12-02:
During GDG Europe Summit, a tinygo workshop was nicely handled by @deadprogram and @conejoninja (thanks a lot guys)

The last step involved MQTT but some environments couldn't compile the code.

The issue is here:
https://github.com/hybridgroup/gdg-2019/issues/6

The code involved here:
https://github.com/hybridgroup/gdg-2019/blob/master/sensor/arduino/step7/main.go

I post it here because I suspect it is something that espat/mqtt driver but feel free to close it if I'm wrong.

deadprogram, 2019-12-18:
Anyone else able to reproduce this one? I have not been able to myself yet.

deadprogram, 2019-12-22:
https://github.com/tinygo-org/tinygo/pull/787 should have fixed that, so I am now closing as that code was merged into TinyGo `dev` branch.

Thank you!

closed 2019-12-22 by deadprogram