tinygo-org/drivers #117: Driver request for DHT-11 and DHT-22

homebrew79, 2020-01-06:
I would like to request a driver for the DHT-11 and DHT-22 Temperature/Humidity sensor. I have seen a few libraries on github for go, but they seem to be specific to raspberry pi.

conejoninja, 2020-01-06:
DHT-11/22 are quite common and it would be good to implement them.

I left this  ( https://github.com/MichaelS11/go-dht ) as a reference for whoever want to take the challenge , this might be a good start since it's a pure go implementation but it depends on "runtime/debug". DHT-xx are timing sensitive and garbage collection might interfere in it.

alankrantas, 2020-03-23:
Tried to write one as an experiment. It...kind of works? But there are still some weird issues. I'll leave this here and see if it can be solved later.

This is based on my micro:bit MakeCode driver (which both DHT11 and DHT22 works). 

I tested this script on micro:bit; only DHT11 works, kind of. The checksum is always a bit off, but humidity and temperature readings looks ok, so I simply ignore checksum here. (The value of 2nd and 4th bytes are very small so I ignore them as well.)

DHT22 doesn't work. All bytes are either 255 or 254 (almost all bits are read HIGH - why?). So I haven't bothered to calculate results for DHT22. On Arduino Uno both DHT11 and DHT22 are like this as well. 

The sensors I have are the 3-pin PCB version, which already pulled high data pins. They performs the same on 3.3V or 5V.

```golang
package main

import (
	"machine"
	"time"
)

type Device struct {
	Pin machine.Pin
}

func New(pin machine.Pin) Device {
	return Device{pin}
}

func (d *Device) Measure() (humidity uint8, temperature uint8, err error) {
	
	var data [40]bool
	var result [5]uint8
	
	// initialize
	d.Pin.Configure(machine.PinConfig{Mode: machine.PinOutput})
	d.Pin.Set(false)
	time.Sleep(time.Millisecond * 18)
	d.Pin.Configure(machine.PinConfig{Mode: machine.PinInput})
	time.Sleep(time.Microsecond * 20)
	
	// wait for ready signal
	for d.Pin.Get() {}
	for !d.Pin.Get() {}
	for d.Pin.Get() {}
	
	// read 40 bits from DHT11
	for i := 0; i < 40; i++ {
	    for d.Pin.Get() {}
	    for !d.Pin.Get() {}
	    time.Sleep(time.Microsecond * 28)
	    data[i] = d.Pin.Get()
	}
	
	// convert bits to numbers
	for i := 0; i < 5; i++ {
	    for j := 0; j < 8; j++ {
	        if data[8 * i + j] {
	            result[i] += 1 << (7 - j)
	        }
	    }
	    // for debug:
	    // print(result[i], " ")
	}
	
	return result[0], result[2], nil
	
}

func main() {
       
       dht := New(machine.P2)
       
       for {
       
            p, t, _ := dht.Measure()
            println("DHT11 humidity =" , p, "/ temperature =", t)
            time.Sleep(time.Millisecond * 5000)
       
       }
       
}
```

aykevl, 2020-03-23:
The protocol appears to be very time sensitive. `time.Sleep` will probably not be precise enough for this purpose (except for the initial 18ms low period). Instead I think it would be better to measure the time between pulses using `time.Now`. Unfortunately that won't work on AVRs yet.

conejoninja, 2020-03-23:
Jut curious, any reason why the first sleep will be precise but not the others?

I do too have timing issues with a sharp gp2y10 (dust sensor), didn't think about time.Now, thanks for the advice.

alankrantas, 2020-03-24:
I changed the bit-counting part to this

```golang
for i := 0; i < 40; i++ {
	    for d.Pin.Get() {}
	    for !d.Pin.Get() {}
	    start := time.Now()
	    for int64(time.Since(start)) <= 28000 {}
	    data[i] = d.Pin.Get()
	}
```

And it works. The checksum matched. The temperature reading looks more accurate. :)

I'm in office right now. Will work on the rest of it later.

aykevl, 2020-03-24:
Reading [the protocol](http://www.ocfreaks.com/basics-interfacing-dht11-dht22-humidity-temperature-sensor-mcu/) it appears that the protocol uses short and long high pulses to send zeroes and ones. So I wonder whether something like this might be a bit more reliable?

```go
for i := 0; i < 40; i++ {
    for !d.Pin.Get() {}           // wait for the 50µs low period to pass
    start := time.Now()           // start time of high pulse
    for d.Pin.Get() {}            // wait for the 26µs-70µs high pulse to pass
    duration := time.Since(start) // time of high pulse
    data[i] = duration > 30*time.Microsecond // setting cutoff at 30µs here
}
```

This more directly measures the duration of the high period.

> Just curious, any reason why the first sleep will be precise but not the others?

Because it is much longer and because it does not seem to be time critical.

alankrantas, 2020-03-24:
@aykevl yes, that's should be the normal way, but due to some reason it's not fast enough. (I did tested with your code but it didn't work.) I had the same problem with micro:bit MakeCode's timing function until I found this solution somewhere online.

So here's the new code. Works for both my DHT11 and DHT22. At least for a while.

_...But_ after a minute or two my micro:bit start to read all zero from both sensors and wouldn't output anything to serial port after reset, unless I flash the code again. I'll try them with other boards later.

```golang
package main

import (
	"machine"
	"time"
)

const (
    DHT11 = iota
    DHT22
)

type Device struct {
	Pin   machine.Pin
	Model uint8
}

func New(pin machine.Pin, model uint8) Device {
	return Device{pin, model}
}

func (d *Device) Measure() (humidity float32, temperature float32, checksum_ok bool, err error) {
	
	var data   [40]bool
	var result [5]uint8
	var h, t   float32
	var cks_ok bool
	
	wait := func (t int64) {
	    start := time.Now()
	    for int64(time.Since(start)) <= t {}
	}
	
	// initialize
	d.Pin.Configure(machine.PinConfig{Mode: machine.PinOutput})
	d.Pin.Set(false)
	wait(18000000)
	d.Pin.Configure(machine.PinConfig{Mode: machine.PinInput})
	wait(20000)
	
	// wait for ready signal
	for d.Pin.Get() {}
	for !d.Pin.Get() {}
	for d.Pin.Get() {}
	
	// read 40 bits from DHT11
	for i := 0; i < 40; i++ {
	    for d.Pin.Get() {}
        for !d.Pin.Get() {}
	    wait(28000)
	    if d.Pin.Get() { data[i] = true }
	}
	
	// convert bits to numbers
	for i := 0; i < 5; i++ {
	    for j := 0; j < 8; j++ {
	        if data[8 * i + j] {
	            result[i] += 1 << (7 - j)
	        }
	    }
	}
	
	// verify checksum
	dta_sum := uint64(result[0] + result[1] + result[2] + result[3])
	if dta_sum >= 512 { dta_sum -= 512 }
	if dta_sum >= 256 { dta_sum -= 256 }
	cks_ok = (dta_sum == uint64(result[4]))
	
	// calculate data
	if d.Model == DHT11 {
	    h = float32(result[0]) + float32(result[1]) / 100
	    t = float32(result[2]) + float32(result[3]) / 100
	    
	} else { // DHT22
	    t_sign := 1
	    if result[2] >= 128 { // negative temperature bit
	        result[2] -= 128
	        t_sign = -1
	    }
	    h = float32(uint16(result[0]) << 8 + uint16(result[1])) / 10
	    t = float32(uint16(result[2]) << 8 + uint16(result[3])) / 10 * float32(t_sign)
	}
	
	return h, t, cks_ok, nil // return data
	
}

func main() {
       
       time.Sleep(time.Millisecond * 100)
       dht := New(machine.P0, DHT11)
       
       for {
       
            p, t, cks, _ := dht.Measure()
            println("Humid =" , p, " Temp =", t, " Checksum ok =", cks)
            time.Sleep(time.Millisecond * 2000)
       
       }
       
}
```

conejoninja, 2020-03-24:
I would define the "wait" function outside Measure(), no need to re-define it each time you want to get a measurement. Also, I would try to avoid float32 as much as possible (in my experience, sometime using floats causes some errors). Returned values from the sensor are usually in the "millis", other temp. sensors return milli degrees instead of degrees, same for humidity, pressure,... Sensors should share interface when possible.

https://github.com/tinygo-org/drivers/blob/master/bmp180/bmp180.go#L83
```go
// ReadTemperature returns the temperature in celsius milli degrees (°C/1000).
func (d *Device) ReadTemperature() (temperature int32, err error) {
```

https://github.com/tinygo-org/drivers/blob/master/bme280/bme280.go#L137
```go
// ReadHumidity returns the relative humidity in hundredths of a percent
func (d *Device) ReadHumidity() (int32, error) {
```

alankrantas, 2020-03-24:
I'm still far from knowing a lot about Go, so there are still room for improvement.

Now the readings are int32, with values = original reading * 100.

I don't want to query the sensors twice, since the required intervals are long, so in this version the users would call Measure() first and get readings after that. (the function name is borrowed from [MicroPython](http://docs.micropython.org/en/latest/esp8266/quickref.html#dht-driver).)

I'm not sure should ReadHumidity() and ReadTemperature() return error as well. 

And just like before, the micro:bit reads all zero after a while.

```golang
package main

import (
    "machine"
    "time"
)

type Device struct {
	Pin machine.Pin
	Model uint8
}

const (
    DHT11 = iota
    DHT22
)

var (
    humidity, temperature int32
    checksum_ok bool
    dht_err error
)

// create new device
func New(pin machine.Pin, model uint8) Device {
	return Device{pin, model}
}

// measure readings
func (d *Device) Measure() (cks bool, err error) {
	var data [40]bool
	var result [5]uint8
	checksum_ok = false
	dht_err = nil
	
	// initialize
	d.Pin.Configure(machine.PinConfig{Mode: machine.PinOutput})
	d.Pin.Set(false)
	wait(18000000)
	d.Pin.Configure(machine.PinConfig{Mode: machine.PinInput})
	wait(20000)
	
	// wait for ready signal
	for d.Pin.Get() {}
	for !d.Pin.Get() {}
	for d.Pin.Get() {}
	
	// read 40 bits from DHT11
	for i := 0; i < 40; i++ {
            for d.Pin.Get() {}
            for !d.Pin.Get() {}
            wait(28000)
            if d.Pin.Get() { data[i] = true }
	}
	
	// convert bits to numbers
	for i := 0; i < 5; i++ {
	    for j := 0; j < 8; j++ {
	        if data[8 * i + j] { result[i] += 1 << (7 - j) }
	    }
	}
	
	// verify checksum
	dta_sum := uint64(result[0] + result[1] + result[2] + result[3])
	if dta_sum >= 512 { dta_sum -= 512 }
	if dta_sum >= 256 { dta_sum -= 256 }
	checksum_ok = (dta_sum == uint64(result[4]))
	
	// calculate data
	if d.Model == DHT11 {
        humidity = int32(result[0]) * 100 + int32(result[1])
        temperature = int32(result[2]) * 100 + int32(result[3])
	    
	} else { // DHT22
        t_sign := int32(1)
        if result[2] >= 128 { // negative temperature bit
            result[2] -= 128
            t_sign = -1
        }
        humidity = (int32(result[0]) << 8 + int32(result[1])) * 10
        temperature = (int32(result[2]) << 8 + int32(result[3])) * 10 * t_sign
	}
	
	if err != nil {
	    dht_err = err
		return
	}
	return checksum_ok, nil // return checksum status
}

func wait(t int64) {
    start := time.Now()
    for int64(time.Since(start)) <= t {}
}

// get humidity reading
func (d *Device) ReadHumidity() (h int32, err error) {
    return humidity, dht_err
}

// get temperature reading
func (d *Device) ReadTemperature() (t int32, err error) {
    return temperature, dht_err
}

// get checksum status
func (d *Device) ReadChecksumOK() (cks bool, err error) {
    return checksum_ok, dht_err
}

func main() {
       dht := New(machine.P0, DHT11)
       for {
            cks, _ := dht.Measure()
            h, _ := dht.ReadHumidity()
            t, _ := dht.ReadTemperature()
            println("H =" , h, "/ T =", t, "/ CKS ok =", cks)
            time.Sleep(time.Millisecond * 2000)
       }
}
```

adrianbn, 2021-04-30:
This probably can be closed since the driver is merged, right?

homebrew79, 2021-04-30:
Closing due to driver merge. Thanks @pleomaxx3002

closed 2021-04-30 by homebrew79