tinygo-org/drivers #125: Request for LSM303AGR (micro:bit v1.5)

alankrantas, 2020-02-04:
The current micro:bit version 1.5 uses a single LSM303AGR accelerometer/magnetometer chip. Might be a close relative of the LSM6DS3 driver you have here? The older 1.3b version uses MMA8653 + MAG3110.

https://tech.microbit.org/hardware/#motion-sensor
https://support.microbit.org/support/solutions/articles/19000087020-micro-bit-motion-sensor-hardware-change

alankrantas, 2020-06-22:
Tried to piece a test code together based on [Adafruit's old LSM303 driver](https://github.com/adafruit/Adafruit_LSM303) (which is pretty basic and the magnetometer registers looks incorrect.).

The following code is confirmed working on my micro:bit v1.5, for both accelerometer and magnetometer. If no one else is available to work on it, I'll later try my best to write a proper-ish driver.

```go
package main

import (
    "machine"
    "time"
)

const (
    
    LSM303_ADDRESS_ACCEL = 0x19
    LSM303_ADDRESS_MAG = 0x1E
    
    LSM303_REGISTER_ACCEL_CTRL_REG1_A = 0x20
    LSM303_REGISTER_ACCEL_OUT_X_L_A = 0x28
    
    LSM303_REGISTER_MAG_MR_REG_M = 0x02
    LSM303_REGISTER_MAG_OUT_X_L_M = 0x68
    
)

func main() {
       
    i2c := machine.I2C0
    i2c.Configure(machine.I2CConfig{})
    
    i2c.WriteRegister(LSM303_ADDRESS_ACCEL, LSM303_REGISTER_ACCEL_CTRL_REG1_A | 0x27, nil)
    i2c.WriteRegister(LSM303_ADDRESS_ACCEL, LSM303_REGISTER_MAG_MR_REG_M | 0x00, nil)
    
    data := make([]byte, 6)
    
    for {

        i2c.ReadRegister(LSM303_ADDRESS_ACCEL, LSM303_REGISTER_ACCEL_OUT_X_L_A | 0x80, data)

        accel_x := int16((uint16(data[1]) << 8) | uint16(data[0]))
        accel_y := int16((uint16(data[3]) << 8) | uint16(data[2]))
        accel_z := int16((uint16(data[5]) << 8) | uint16(data[4]))

        i2c.ReadRegister(LSM303_ADDRESS_MAG, LSM303_REGISTER_MAG_OUT_X_L_M, data)

        mag_x := int16((uint16(data[1]) << 8) | uint16(data[0]))
        mag_y := int16((uint16(data[3]) << 8) | uint16(data[2]))
        mag_z := int16((uint16(data[5]) << 8) | uint16(data[4]))

	println("LSM303_ACCR:", accel_x, accel_y, accel_z)
	println("LSM303_MAG:", mag_x, mag_y, mag_z)
        println("-----")

        time.Sleep(time.Millisecond * 100)

    }
       
}
```

Output:

> LSM303_ACCR: -64 4608 -15680
> LSM303_MAG: -262 193 79
> -----
> LSM303_ACCR: 0 4672 -15680
> LSM303_MAG: -261 187 81
> -----
> LSM303_ACCR: -64 4608 -15680
> LSM303_MAG: -256 192 81
> -----
> LSM303_ACCR: -64 4480 -15616
> LSM303_MAG: -263 194 84
> -----
> LSM303_ACCR: -64 4608 -15616
> LSM303_MAG: -261 190 85
> -----
> LSM303_ACCR: 0 4480 -15616
> LSM303_MAG: -250 192 80
> -----

conejoninja, 2020-06-22:
This will be a great addition to the driver's list, I do not have one of the newest micro:bit, so feel free to continue the work and make a PR of it.

alankrantas, 2020-07-02:
I'll close this since in the end I solved my own request.

closed 2020-07-02 by alankrantas