tinygo-org/drivers #148: Unable to build ultrasonic sensor distance reading code

souvikhaldar, 2020-04-25:
I wrote this code https://play.golang.org/p/DimqElkkaaW
I've installed the required driver (using go get tinygo.org/x/drivers) but while building it using tinygo build -o dist main.go I'm getting the following error:
```
/usr/bin/ld: /tmp/tinygo498204373/main.o: in function `(machine.Pin).Configure':
../../../../usr/local/tinygo/src/machine/machine_generic.go:23: undefined reference to `__tinygo_gpio_configure'
/usr/bin/ld: /tmp/tinygo498204373/main.o: in function `(machine.Pin).Set':
../../../../usr/local/tinygo/src/machine/machine_generic.go:27: undefined reference to `__tinygo_gpio_set'
/usr/bin/ld: ../../../../usr/local/tinygo/src/machine/machine_generic.go:27: undefined reference to `__tinygo_gpio_set'
/usr/bin/ld: /tmp/tinygo498204373/main.o: in function `(machine.Pin).Get':
../../../../usr/local/tinygo/src/machine/machine_generic.go:31: undefined reference to `__tinygo_gpio_get'
/usr/bin/ld: /tmp/tinygo498204373/main.o: in function `(machine.Pin).Set':
../../../../usr/local/tinygo/src/machine/machine_generic.go:27: undefined reference to `__tinygo_gpio_set'
/usr/bin/ld: /tmp/tinygo498204373/main.o: in function `(machine.Pin).Get':
../../../../usr/local/tinygo/src/machine/machine_generic.go:31: undefined reference to `__tinygo_gpio_get'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo498204373/main: exit status 1
```

souvikhaldar, 2020-04-25:
When I tried to flash it to **Arduino Uno** board using `tinygo flash -target=arduino -port=/dev/ttyACM0 main.go`, I got the following error:  
```

souvikhaldar, 2020-04-25:
#reopen

souvikhaldar, 2020-04-25:
Reverted to old version, code looks like- https://play.golang.org/p/gWwj8nLiepM
I provided the target and built again using target=arduino now I'm getting this error:
tinygo build -target=arduino -o dist main.go 
/tmp/tinygo706642821/main.o: In function `LBB22_28':
main.go:(.text.time.Since+0x554): undefined reference to `__divmoddi4'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo706642821/main.o: In function `LBB23_14':
main.go:(.text.main.go.main+0x1ce): undefined reference to `__divmoddi4'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo706642821/main: exit status 1

conejoninja, 2020-04-26:
As per the slack thread, this issue is caused by a bug on AVR backend that is fixed upstream. It will eventually land on tinygo. It's not related to the driver mentioned. I'm closing this issue.

closed 2020-04-26 by conejoninja

aykevl, 2020-04-26:
Did you test TinyGo version 0.13.0 or 0.13.1? It may have fixed this issue with https://github.com/tinygo-org/tinygo/pull/1023.

If that doesn't help, you can also try https://github.com/tinygo-org/tinygo/pull/1056 (build with `go install -tags=llvm11`). It should fix the issue with `__divmoddi4`.

souvikhaldar, 2020-04-27:
Let me try it and report here.

souvikhaldar, 2020-04-27:
> As per the slack thread, this issue is caused by a bug on AVR backend that is fixed upstream. It will eventually land on tinygo. It's not related to the driver mentioned. I'm closing this issue.

Please reopen until I report my findings here, should be a good document to refer to later.

AllAwesome497, 2020-05-31:
I'm also having the same issue... I tried both of @aykevl's solutions, but neither of them work...

<details>
    <summary>Error I get:</summary>
```bash
allawesome497@mintOS:~/Desktop/repl.it/tinygo$ make shell
/tmp/tinygo252264819/main.o: In function `LBB19_26':
main.go:(.text.main.go.main+0x5cc): undefined reference to `__divmoddi4'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo252264819/main.o: In function `LBB19_92':
main.go:(.text.main.go.main+0xfe2): undefined reference to `__divmoddi4'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo252264819/main: exit status 1
Makefile:9: recipe for target 'main' failed
make: *** [main] Error 1
```
</details>

I have everything-11 installed.

<details>
    <summary>Output of `ldd $GOPATH/bin/tinygo`:</summary>
```bash
allawesome497@mintOS:~$ ldd $GOPATH/bin/tinygo
	linux-vdso.so.1 (0x00007ffe119f3000)
	libLLVM-10.so.1 => /usr/lib/x86_64-linux-gnu/libLLVM-10.so.1 (0x00007fef47a20000)
	libclang-10.so.1 => /usr/lib/x86_64-linux-gnu/libclang-10.so.1 (0x00007fef45a30000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fef45810000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fef45480000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fef450e0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fef44ec8000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fef44ad0000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007fef448c8000)
	libedit.so.2 => /usr/lib/x86_64-linux-gnu/libedit.so.2 (0x00007fef44690000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fef44470000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fef44268000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fef44060000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007fef4c3e8000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fef4c210000)
	libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007fef43e30000)

```
</details>

<details>
    (note that shell/main.go and main.go are different scripts, shell.go is to read the serial data from the serial connection.)
    <summary>My code:</summary>
    <details>
        <summary>main.go</summary>
```go

package main

import (
    "time"

    "machine"
)

func main() {

    in := machine.D53

    output := machine.PinConfig{
        Mode: machine.PinOutput,
    }
    input := machine.PinConfig{
        Mode: machine.PinInput,
    }

    for {
        in.Configure(output)
        startTime := time.Now()
        in.Low()
        time.Sleep(time.Microsecond * 2)
        in.High()
        time.Sleep(10 * time.Microsecond)
        in.Low()
        in.Configure(input)
        for !in.Get() {
        }
        endTime := time.Now()
        diff := endTime.Sub(startTime)
        println(diff.Microseconds())
    }
}
```
</details>
<details>
    <summary>`output/shell/main.go`</summary>
```go
package main

import (
	"fmt"
	"os"
	"os/signal"
	"time"

	"github.com/tarm/serial"
)

func main() {
	sig := make(chan os.Signal, 1)
	signal.Notify(sig, os.Interrupt, os.Kill)

	conf := &serial.Config{
		Name:        "/dev/ttyACM0",
		Baud:        115200,
		ReadTimeout: time.Second,
	}

	s, err := serial.OpenPort(conf)

	if err != nil {
		panic(err)
	}
	defer fmt.Println("Closing :)")
	defer s.Close()
	fmt.Println("OK :D")
	for {
		buf := make([]byte, 128)
		_, err := s.Read(buf)
		if err != nil {
			fmt.Println("oof... Error", err.Error(), string(buf), buf)
		}

		fmt.Print(string(buf))

		select {
		case <-sig:
			return
		default:
		}
	}
}
```
</details>
<details>
    <summary>My makefile:</summary>
```make
export PORT=/dev/ttyACM0

.PHONY: shell
shell: main
	busybox microcom $(PORT)
.PHONY: main

main: ./*
	@sudo chmod a+rw $(PORT)
	@tinygo flash -target=arduino-mega2560 -port "$(PORT)" -ocd-output main.go
```
</details>
</details>
```bash
allawesome497@mintOS:~$ tinygo
No command-line arguments supplied.
TinyGo is a Go compiler for small places.
version: 0.13.1
```
<details>
    <summary>Output of `go env`</summary>
```bash
allawesome497@mintOS:~$ go env
GO111MODULE=""
GOARCH="amd64"
GOBIN="/home/allawesome497/go//bin"
GOCACHE="/home/allawesome497/.cache/go-build"
GOENV="/home/allawesome497/.config/go/env"
GOEXE=""
GOFLAGS="-tags=arduino_mega2560"
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOINSECURE=""
GONOPROXY=""
GONOSUMDB=""
GOOS="linux"
GOPATH="/home/allawesome497/go/"
GOPRIVATE=""
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/usr/local/go"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/usr/local/go/pkg/tool/linux_amd64"
GCCGO="gccgo"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD=""
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build256783623=/tmp/go-build -gno-record-gcc-switches"
```
</details>
<details>
    <summary>Output of `tinygo env`</summary>
```bash
allawesome497@mintOS:~$ tinygo env
GOOS="linux"
GOARCH="amd64"
GOROOT="/usr/local/go/"
GOPATH="/home/allawesome497/go/"
GOCACHE="/home/allawesome497/.cache/tinygo"
CGO_ENABLED="1"
TINYGOROOT="/tmp/tinygo"
```
</details>

aykevl, 2020-06-01:
This is not a bug in TinyGo but in LLVM. I've fixed the bug upstream but it'll take a while (until around September or so) before the next LLVM release.

The `ldd` output shows you are still using LLVM 10. Unfortunately the early LLVM support for TinyGo I added is now broken with other changes so can't be used as-is.

AllAwesome497, 2020-06-02:
Alright, thanks. Theres also a few issues w/ arudino suport (mainly the fact that we cant get time in any way (even the arduino millis() func), and gourotines wont start due to not having a scheduler)  but ill make an issue on the main repo for those. Thanks for the response.