tinygo-org/drivers #149: ws2812 method WriteByte already declared

clholzin, 2020-05-03:
I'm receiving an error when flashing for Curcuit Playground Express the ws2812 driver

method WriteByte already declared

tinygo flash -target=circuitplay-express ./main.go
`# tinygo.org/x/drivers/ws2812
../../tinygo.org/x/drivers/ws2812/ws2812_m0_48m.go:13:17: method WriteByte already declared for type Device struct{Pin machine.Pin}
../../tinygo.org/x/drivers/ws2812/ws2812_generic.go:10:17: 	other declaration of WriteByte
`

Is there a flag i'm missing for this platform? 
I'm open to making a PR if provided some direction should a change be needed.

clholzin, 2020-05-03:
Commenting out generic fix my issue.

closed 2020-05-03 by clholzin

aykevl, 2020-05-03:
That's strange, ws2812_generic.go should have been excluded from the build because of the `// +build !baremetal`. And in fact this is tested in CI, so I don't understand how this could go wrong.

This might be helpful:

```
$ tinygo version
$ tinygo info -target=circuitplay-express
```