tinygo-org/drivers #151: SPI not declared by package machine

ArtemkaKun, 2020-05-16:
Hi,
I have an SSD1306 OLED 128x32 display. When I try to use an example for SSD1306 driver and build, have 2 errors:
`../../../../root/go/src/tinygo.org/x/drivers/ssd1306/ssd1306.go:38:19: SPI not declared by package machine
../../../../root/go/src/tinygo.org/x/drivers/ssd1306/ssd1306.go:63:25: SPI not declared by package machine
`
Any thoughts on what can be wrong? SPI type actually declared in machine package :/

conejoninja, 2020-05-16:
Hello, which command are you using to build? Looks like the target is missing.

ArtemkaKun, 2020-05-16:
> Hello, which command are you using to build? Looks like the target is missing.

Hi. I have an Arduino Nano, so I use `tinygo flash -target=arduino-nano  -port /dev/ttyUSB0 main.go`
(main.go it's just a simple go file that contains code from SSD1306 example)

conejoninja, 2020-05-16:
Unfortunately, SPI support is missing for the Nano :( 

https://tinygo.org/microcontrollers/arduino-nano/

ArtemkaKun, 2020-05-16:
> Unfortunately, SPI support is missing for the Nano :(
> 
> https://tinygo.org/microcontrollers/arduino-nano/

Oh, sorry for my incompetence. I try to connect to my display with I2C.
Here is the code
`func main() {
	machine.I2C0.Configure(machine.I2CConfig{
		Frequency: machine.TWI_FREQ_400KHZ,
	})

	display := ssd1306.NewI2C(machine.I2C0)
	display.Configure(ssd1306.Config{
		Address: ssd1306.Address_128_32,
		Width:   128,
		Height:  32,
	})

	display.ClearDisplay()

	x := int16(0)
	y := int16(0)
	deltaX := int16(1)
	deltaY := int16(1)
	for {
		pixel := display.GetPixel(x, y)
		c := color.RGBA{255, 255, 255, 255}
		if pixel {
			c = color.RGBA{0, 0, 0, 255}
		}
		display.SetPixel(x, y, c)
		display.Display()

		x += deltaX
		y += deltaY

		if x == 0 || x == 127 {
			deltaX = -deltaX
		}

		if y == 0 || y == 31 {
			deltaY = -deltaY
		}
		time.Sleep(1 * time.Millisecond)
	}`

conejoninja, 2020-05-16:
Oh, right. The drivers works for I²C and SPI, but it need both declared to work properly. Not sure if using some build tags we could split the driver in two and only load the version needed. Other solution is creating SSD1306SPI and SSD1306I2C drivers, but I prefer not to /cc @aykevl

ArtemkaKun, 2020-05-16:
> Oh, right. The drivers works for I²C and SPI, but it need both declared to work properly. Not sure if using some build tags we could split the driver in two and only load the version needed. Other solution is creating SSD1306SPI and SSD1306I2C drivers, but I prefer not to /cc @aykevl

Ok, thank you :) Try to use another way :/
Issue closed.

conejoninja, 2020-05-16:
I'm re-opening this issue. This is a problem with the driver and should get fixed or at least propose an alternative, I think some other drivers also have I2C+SPi versions and there are other boards with only SPI or I2C support. For the moment, you could remove the references to SPI from the driver and it should work.

ArtemkaKun, 2020-05-16:
> I'm re-opening this issue, thanks for bringing this up. This is a problem with the driver and should get fixed or at least propose an alternative, I think some other drivers also have I2C+SPi versions and there are other boards with only SPI or I2C support. For the moment, you could remove the references to SPI from the driver and it should work.

So, I removed SPI parts, as you say, and it works great. Actually I see only two solutions to the problem:
1) Comment SPI code until SPI will be supported by TinyGo on Arduino Uno.
2) Separate SPI and I2C implementation.

aykevl, 2020-05-16:
Longer term, I think the best solution would be to create `machine.SPI` and `machine.I2C` interfaces which are used by all the drivers. This has a few advantages:

1. The drivers are no longer coupled with the machine package, or at least not as tightly. This may allow re-use of the drivers in regular Go.
2. It may make it possible to do some unit testing without requiring any real hardware (using mocks/fakes).
3. It allows using these drivers on chips that have one of the interfaces (I2C, SPI) not implemented. I hadn't originally thought of this issue but using interfaces would nicely work around that as well.

TinHead, 2020-08-01:
> So, I removed SPI parts, as you say, and it works great. Actually I see only two solutions to the problem:
>
> 1. Comment SPI code until SPI will be supported by TinyGo on Arduino Uno.
> 2. Separate SPI [...]

Hi @ArtemkaKun just out of curiosity did you get the display to work?

ArtemkaKun, 2020-08-01:
> > > I'm re-opening this issue, thanks for bringing this up. This is a problem with the driver and should get fixed or at least propose an alternative, I think some other drivers also have I2C+SPi versions and there are other boards with only SPI or I2C support. For the moment, you could remove the references to SPI from the driver and it should work.
> > 
> > 
> > So, I removed SPI parts, as you say, and it works great. Actually I see only two solutions to the problem:
> > 
> > 1. Comment SPI code until SPI will be supported by TinyGo on Arduino Uno.
> > 2. Separate SPI and I2C implementation.
> 
> Hi @ArtemkaKun just out of curiosity did you get the display to work after removing the SPI calls? I've been trying for a while with a 128x64 SSD1306 and all I get is a garbled display.

Hi. Actually I only achieved compilation without errors :/ I don't test if the display can show something. Give me a sec.

ArtemkaKun, 2020-08-01:
@TinHead can you show your code?

TinHead, 2020-08-06:
> @TinHead can you show your code?

yup you can use my fork here: https://github.com/TinHead/drivers 
just do:
`go get -d github.com/TinHead/drivers/ `
and use the example in the examples folder