tinygo-org/drivers #152: Wifinina: Needs TLS Support

seanaye, 2020-05-18:
As far as I can tell there is no support for TCP TLS connections for wifinina

```go
        const server = "google.com:80"
	println("\r\n---------------\r\nDialing TCP connection")
	conn, err = net.Dial("tcp", server)
	for ; err != nil; conn, err = net.Dial("tcp", server) {
		println("connection failed: " + err.Error())
		time.Sleep(5 * time.Second)
	}
	println("Connected!\r")
```
This prints `Connected!` and behaves as expected

```go
        const server = "google.com:443"
	println("\r\n---------------\r\nDialing TCP connection")
	conn, err = tls.Dial("tcp", server, &tls.Config{})
	for ; err != nil; conn, err = tls.Dial("tcp", server, &tls.Config{}) {
		println("connection failed: " + err.Error())
		time.Sleep(5 * time.Second)
	}
	println("Connected!\r")
```
this prints
```
connection failed: wifinina error: 0x01
connection failed: wifinina error: 0x03
connection failed: wifinina error: 0x03
...
```

I'm not sure why the first response is a different error code.

My board is the Arduino Nano 33 IoT

I'm going off [Dial](https://github.com/tinygo-org/drivers/blob/c12f78eb50d1f49e03d98568e95489c68c089aa4/net/tls/tls.go#L14)

I've traced the issue back to [here](https://github.com/tinygo-org/drivers/blob/9f23761c5e8395752f30968896c33f3d8a90259a/wifinina/wifinina.go#L158) I see that TCP has a value of "iota" but I have no idea why, so I'm not sure what the TLS value should be.

I would be happy to contribute if someone can explain this to me.

deadprogram, 2020-05-18:
The implementation for TLS is already done. See https://github.com/tinygo-org/drivers/blob/master/wifinina/tcp.go#L40 and also https://github.com/tinygo-org/drivers/blob/master/net/tls/tls.go

Hope that helps!

seanaye, 2020-05-18:
@deadprogram Thanks for the reply.

Yes I am using tls.Dial this always results in timeout. In my code sample above net.Dial works fine but tls.Dial does not. Maybe it is an issue with my board.

deadprogram, 2020-05-18:
I think these are the wifinina internal error codes: https://github.com/tinygo-org/drivers/blob/08cc84ce48bffe07cde78c688cd80249d179cec1/wifinina/wifinina.go#L137-L153

@bgould can confirm this.

So at a quick glance, seems like you might need to wait a little bit of time before your first attmpt to connect. Also, since the TLS connection takes a lot longer to establish, you might need to wait longer between retries.

seanaye, 2020-05-18:
Yes these are the wifinina error codes. I've increased the wait time to 30 seconds and still receive the same error

```go
	println("\r\n---------------\r\nDialing TCP connection")
	conn, err = tls.Dial("tcp", server, conf)
	for ; err != nil; conn, err = tls.Dial("tcp", server, conf) {
		println("connection failed: " + err.Error())
		time.Sleep(30 * time.Second)
	}
	println("Connected!\r")
```

Very strange, the standard port 80 non-TLS connection happens very quickly

bgould, 2020-05-19:
Its true, TLS does not work.  I can get it working however... I've also been trying to get UDP working as well first but I seem to be stuck on something with that so I can take a break and see about TLS.  I'm also in the middle of refactoring the low-level protocol implementation so that it is easier to understand.  I'll be opening a PR with these changes soon and I'll try to include a fix for TLS with that or shortly after.

seanaye, 2020-05-19:
@bgould let me know if/how I can help as this is something I need for a project

torntrousers, 2020-08-09:
TLS would be really useful, lots of sites no longer support http, only https. Any update on this?

bgould, 2020-08-14:
I have been able to figure out how to implement this, it is a bit rough right now but I'll have it cleaned up soon, should be ready in plenty of time for the next release

Nerzal, 2021-02-25:
@bgould do you have any updates on this?

deadprogram, 2021-05-12:
This is in the latest release, so now closing. Thanks!

closed 2021-05-12 by deadprogram