tinygo-org/drivers #155: SPI sdcard driver

sago35, 2020-05-24:
I want to read and write sdcard with SPI, do you have an implementation?

conejoninja, 2020-05-24:
I'm not sure how the sdcard is implemented in hardware, but there's a driver for FLASH memories ( https://github.com/tinygo-org/drivers/tree/master/flash ) that might be worth take a look at.

Some work on the filesystems , thanks to @bgould that I'm sure will be helpful too
https://github.com/bgould/tinyfs
https://github.com/bgould/go-fatfs
https://github.com/bgould/go-littlefs

aykevl, 2020-05-24:
I investigated SD cards a while ago. There are two ways to drive them:

1. Using a SPI based protocol. Slow, but very portable.
2. Using some custom SPI-like protocol that is much faster, but seems to be covered in patens or something, I don't know. The last time I looked at this, I decided not to go down that route.

I think the protocol for flash chips is different from the protocol for SPI. You can take a look here for how the protocol works: https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py

bgould, 2020-05-25:
The SPI flash driver won't be useful for SD card but the fatfs filesystem in tinyfs will be since for the most part all use cases of interest probably would use FAT-formatted cards.

The facts that Ayke stated about the protocol are correct as far as I know.  I would just add that I think the command set and packet structure for both protocols is mostly the same, except that using the faster protocol (I don't know the actual name of that protocol, I'm just going to refer to it as the "SD protocol" which is very likely wrong) requires CRC16 checksums on its packets, and I think the SPI protocol has a couple of extra commands that aren't relevant to the SD protocol.  This page is a good resource on using the SPI mode of operation, and also happens to be published by the author of the FatFs library that is integrated into tinyfs: http://elm-chan.org/docs/mmc/mmc_e.html

Adding support for reading from SD cards using the SPI protocol is certainly achievable, and mostly would involve writing a low-level driver for reading and writing blocks according to this interface which Ayke proposed for the SPI NOR flash driver: https://github.com/bgould/tinyfs/blob/e8761ce4bbe46691c366c4c330373de9e69035f3/tinyfs.go#L43

With the block device driver it will be simple to use the fatfs implementation in tinyfs to work with a read-only FAT filesystem on the card.  It would essentially be the same as this example, except that the SD block device driver would be used instead of the SPI flash block device driver: https://github.com/bgould/tinyfs/blob/master/examples/console/fatfs/spi/main.go#L21 .  

Now that https://github.com/tinygo-org/tinygo/pull/1012 is merged, since tinyfs implements that interface it would then be possible mount the FAT filesystem on the SD card and access it using the os.File abstraction, which would be pretty nice and easy to use I think.

BTW the main reason that the filesystem would need to be read-only is because I've set the read-only flag in the C header for the FatFs library because for NOR flash memory I felt it was necessary to first create a block caching layer to minimize the number of erase operations, otherwise it would be too easy to quickly wear out commonly used blocks on a device with a FAT filesystem (this problem does not exist with littlefs so that implementation supports read-write).  I didn't try implement such a thing yet so I just disabled read-write for FAT for the time being.  That restriction could be irrelevant with SD cards because I think they might do wear-leveling/caching internally, and if so it would be easy enough to enable read-write for FAT in tinyfs.  In any case, it is significantly useful even if just read-only operations can be implemented for now... it would mean that files could be written to the card using a regular computer and then read from a TinyGo program on a microcontroller (could be useful for fonts/graphics, wifi credentials, etc).

SAMD51 chips have a hardware host controller for the SD protocol in 4-bit transfer mode at least according to the datasheet (SPI mode only supports 1-bit transfers, since that is how SPI works).  Once SD card support in SPI mode is working it could be possible to try to get the hardware peripheral working as well for faster operation.  

tl;dr is that a driver would need to be implemented for low-level block operations using the SD command set over SPI.

Personally I'm busy with trying to improve the wifinina driver right now so I cannot work on this myself yet, but I am interested in this feature and if someone were to do it I would be glad to test it or make/review any changes to tinyfs/fatfs to accomodate it.  Otherwise I may have some time to try implementing a SD driver at some point down the road... but right now it is not at the top of my list of things to get done.

sago35, 2020-05-25:
Thank you for your comment.

For now, I'm going to start by porting sdcard.py first.
After that I will build the driver to satisfy the tinyfs.BlockDevice interface.

AllAwesome497, 2020-06-24:
You could also look at [arduino's libraries](https://github.com/arduino-libraries/SD/) for this, they also include the fat32 file system

sago35, 2020-06-24:
Thanks for the information.
I was looking at the same place.

So far, I've been able to read and write at a minimum with the following as well.
https://github.com/arduino-libraries/SD/blob/master/src/utility/Sd2Card.cpp

Right now I'm writing the driver for the RTL8720 (wifi + ble) and will be working on the sdcard driver after that.

AllAwesome497, 2020-06-24:
Gotcha. I recently got some sd card readers, sadly I dont have the time to make the lib myself atm, so im happy someone else is doing it.

AllAwesome497, 2020-06-25:
@sago35 you mentioned having the ability to read and files from the sd card, what do you mean by that? If I'm able to get an io.ReadSeeker from a file on the card could you send what you have now?

sago35, 2020-06-25:
@AllAwesome497 
At the moment, it is only primitive reading and writing.
And at this point in time, the quality is poor, and for some reason, it's getting an error or normal.

* http://elm-chan.org/docs/mmc/mmc_e.html
    * CMD17 : READ_SINGLE_BLOCK
    * CMD24 : WRITE_BLOCK

The source code as of now is below.
It's probably not useful.

https://gist.github.com/sago35/c9a9d2c96279fbcbaf8d09cf0ec4a113

sago35, 2020-07-09:
Hello @bgould 

The prototype is ready.
I need to make a few changes to the tinyfs example, but so far it seems to be working correctly.

https://github.com/sago35/tinyfs/commits/sdcard

bgould, 2020-07-11:
@sago35 this is awesome thank you.  I will try it out with hardware today or tomorrow, so far it looks great though

closed 2022-08-16 by deadprogram