tinygo-org/drivers #179: Conflicting info regarding Nano33 IoT in drivers/espat and drivers/wifinina READMEs

s-video, 2020-08-01:
Hello,

At the bottom of the [tinygo.org page for the Arduino Nano33 IoT](https://tinygo.org/microcontrollers/arduino-nano33-iot/) is a note that says:

>For information on how to use the built-in NINA-W102 wireless chip, please see the “espat” driver in the TinyGo drivers repository located at https://github.com/tinygo-org/drivers/tree/master/espat.

This links to the drivers/espat directory, whose README says:

>In order to install the needed firmware on the Arduino Nano33 IoT board's built-in NINA W102 chip, you will need to use the arduino-nano33-iot branch of this fork of the firmware

And some instructions for flashing the NINA W102 chip.

However, the README in the [drivers/wifinina directory](https://github.com/tinygo-org/drivers/tree/master/wifinina) states

>PLEASE NOTE: New Arduino Nano33 IoT boards already have the WiFiNINA firmware pre-installed, so you should not need to install the firmware yourself.

So, it's unclear if one needs to flash the NINA W102 chip on a Nano33 IoT themselves or not. From what I can guess, the README in the drivers/espat directory might be saying that one would need to flash the NINA W102 to use the drivers/espat driver _specifically_ but not if they wish to use the drivers/wifinina driver. If that's the case, I think the note at the tinygo.org page can potentially cause a great amount of confusion and wasted time, as it's probably the first page someone would read if they'd like to use a Nano33 IoT with tinygo, before they would explore the drivers directory and see the drivers/wifinina README and realize they don't need to do extra legwork to connect their device to the internet. (I'm typing all of this hypothetically, but I just sunk a few hours dealing with this scenario.)

bgould, 2020-08-07:
The issue stems from the fact that the `espat` driver was created before the `wifinina` one, so at the time that note on the website was written, it was accurate and indeed it required flashing new firmware onto the ESP32 module.

Since then, the `wifinina` driver was created and for the Nano 33 IOT probably is a better starting place, so perhaps the website should be updated.  The website is in a different repo, the change will need to be made here: https://github.com/tinygo-org/tinygo-site/blob/release/content/microcontrollers/arduino-nano33-iot.md

deadprogram, 2020-09-16:
This has been addressed here https://github.com/tinygo-org/tinygo-site/pull/114/commits/8e5326d42090a8f442fe6ed21c3da93a8c7180d0 so now closing. Thanks everyone!

closed 2020-09-16 by deadprogram