tinygo-org/drivers #182: `make test` is failing for amg88xx

timboldt, 2020-08-11:
$ make test
tinygo build -size short -o ./build/test.hex -target=itsybitsy-m0 ./examples/adt7410/main.go
   code    data     bss |   flash     ram
  44300    1420    4312 |   45720    5732
181cbf93b088c95d6ef403448eb9bb21  ./build/test.hex
tinygo build -size short -o ./build/test.hex -target=itsybitsy-m0 ./examples/adxl345/main.go
   code    data     bss |   flash     ram
  10868      40    4312 |   10908    4352
cb577413af124494230028953476610b  ./build/test.hex
tinygo build -size short -o ./build/test.hex -target=pybadge ./examples/amg88xx
# ./examples/amg88xx
examples/amg88xx/main.go:16:22: SPI1_SDO_PIN not declared by package machine
examples/amg88xx/main.go:16:3: unknown field SDO in struct literal
examples/amg88xx/main.go:17:22: SPI1_SDI_PIN not declared by package machine
examples/amg88xx/main.go:17:3: unknown field SDI in struct literal
make: *** [Makefile:16: smoke-test] Error 1

timboldt, 2020-08-11:
This is a probable interaction between a copy of tiny-go/drivers retrieved from `go get` and the fork of tiny-go/drivers that I was trying to test.

After deleting the copy that Go downloaded, `make test` just fails outright immediately. I assume there is a Go Modules trick to get Go to use the code in the current directory?

conejoninja, 2020-08-11:
Are you using the latest version of tinygo? SPI1_SDI_PIN & SPI1_SDO_PIN were introduced here : https://github.com/tinygo-org/tinygo/commit/d1c4ed664e540e00b3d18ff340d6798c45b382cf

timboldt, 2020-08-11:
Ahah, that was it. Upgrading from 0.13.x to 0.14.0 resolved the issue.

closed 2020-08-11 by timboldt