tinygo-org/drivers #185: GPS UART example build error for Arduino Nano

tomprouseau, 2020-08-17:
Hello,

I've copied the example for interfacing with a GPS using the UART option. I had to change UART1 in the example to UART0 as the target device is an arduino nano. When I run:

`tinygo build -target=arduino -o gps.bin main.go`

I get the following error: 

tinygo: /home/circleci/project/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:9794: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function &): Assertion `InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"' failed.
SIGABRT: abort
PC=0x7f009d35b18b m=4 sigcode=18446744073709551610

goroutine 0 [idle]:
runtime: unknown pc 0x7f009d35b18b
stack: frame={sp:0x7f0075e8c540, fp:0x0} stack=[0x7f007568e248,0x7f0075e8de48)
00007f0075e8c440:  0000000000000006  00000000063bed11 
00007f0075e8c450:  0000000000002642  00000000063bc1c9 
00007f0075e8c460:  00000000063bed55  0000000000000001 
00007f0075e8c470:  0000000000000002  0000000000000000 
00007f0075e8c480:  0000000000000000  0000000000000000 
00007f0075e8c490:  0000000000000000  0000ffff00001fa0 
00007f0075e8c4a0:  0000000400000000  0000000000000000 
00007f0075e8c4b0:  0000000000000000  00007f0060000020 
00007f0075e8c4c0:  00007f0044a54d70  0000000000000122 
00007f0075e8c4d0:  0000000000000122  00007f0044a54d60 
00007f0075e8c4e0:  0000000000000140  00007f009d3b32d6 
00007f0075e8c4f0:  3439373975e8c550  38b2337b668e6200 
00007f0075e8c500:  00007f00609f4180  00007f0075e8c7a0 
00007f0075e8c510:  0000000000000121  0000000000000121 
00007f0075e8c520:  0000000000000122  00007f0044a54d70 
00007f0075e8c530:  00007f0075e8c550  00007f009d3a3d89 
00007f0075e8c540: <0000000000000000  00007f009d3b2850 
00007f0075e8c550:  00000000fbad8000  00007f0044a54d70 
00007f0075e8c560:  00007f0044a54dd5  00007f0044a54d70 
00007f0075e8c570:  00007f0044a54d70  00007f0044a54e91 
00007f0075e8c580:  00007f0044a54e9c  00007f0044a54d70 
00007f0075e8c590:  00007f0044a54e9c  0000000000000000 
00007f0075e8c5a0:  0000000000000000  0000000000000000 
00007f0075e8c5b0:  0000000000000000  0000000000000000 
00007f0075e8c5c0:  fffffffe7fffffff  ffffffffffffffff 
00007f0075e8c5d0:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c5e0:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c5f0:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c600:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c610:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c620:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c630:  ffffffffffffffff  ffffffffffffffff 
runtime: unknown pc 0x7f009d35b18b
stack: frame={sp:0x7f0075e8c540, fp:0x0} stack=[0x7f007568e248,0x7f0075e8de48)
00007f0075e8c440:  0000000000000006  00000000063bed11 
00007f0075e8c450:  0000000000002642  00000000063bc1c9 
00007f0075e8c460:  00000000063bed55  0000000000000001 
00007f0075e8c470:  0000000000000002  0000000000000000 
00007f0075e8c480:  0000000000000000  0000000000000000 
00007f0075e8c490:  0000000000000000  0000ffff00001fa0 
00007f0075e8c4a0:  0000000400000000  0000000000000000 
00007f0075e8c4b0:  0000000000000000  00007f0060000020 
00007f0075e8c4c0:  00007f0044a54d70  0000000000000122 
00007f0075e8c4d0:  0000000000000122  00007f0044a54d60 
00007f0075e8c4e0:  0000000000000140  00007f009d3b32d6 
00007f0075e8c4f0:  3439373975e8c550  38b2337b668e6200 
00007f0075e8c500:  00007f00609f4180  00007f0075e8c7a0 
00007f0075e8c510:  0000000000000121  0000000000000121 
00007f0075e8c520:  0000000000000122  00007f0044a54d70 
00007f0075e8c530:  00007f0075e8c550  00007f009d3a3d89 
00007f0075e8c540: <0000000000000000  00007f009d3b2850 
00007f0075e8c550:  00000000fbad8000  00007f0044a54d70 
00007f0075e8c560:  00007f0044a54dd5  00007f0044a54d70 
00007f0075e8c570:  00007f0044a54d70  00007f0044a54e91 
00007f0075e8c580:  00007f0044a54e9c  00007f0044a54d70 
00007f0075e8c590:  00007f0044a54e9c  0000000000000000 
00007f0075e8c5a0:  0000000000000000  0000000000000000 
00007f0075e8c5b0:  0000000000000000  0000000000000000 
00007f0075e8c5c0:  fffffffe7fffffff  ffffffffffffffff 
00007f0075e8c5d0:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c5e0:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c5f0:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c600:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c610:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c620:  ffffffffffffffff  ffffffffffffffff 
00007f0075e8c630:  ffffffffffffffff  ffffffffffffffff 

goroutine 1 [syscall]:
runtime.cgocall(0xd80fa0, 0xc0023155a8, 0xc0059daa40)
        /usr/local/go/src/runtime/cgocall.go:133 +0x5b fp=0xc002315578 sp=0xc002315540 pc=0x98604b
tinygo.org/x/go-llvm._Cfunc_LLVMTargetMachineEmitToMemoryBuffer(0x8789220, 0x87a6990, 0x1, 0xc0059daa38, 0xc0059daa40, 0xc000000000)
        _cgo_gotypes.go:9409 +0x4d fp=0xc0023155a8 sp=0xc002315578 pc=0xc284bd
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer.func1(0x8789220, 0x87a6990, 0xc000000001, 0xc0059daa38, 0xc0059daa40, 0x1b)
        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x111 fp=0xc0023155e8 sp=0xc0023155a8 pc=0xc51391
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer(0x8789220, 0x87a6990, 0x1, 0xc0059e4660, 0x1b, 0x14)
        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x84 fp=0xc002315650 sp=0xc0023155e8 pc=0xc376a4
github.com/tinygo-org/tinygo/builder.Build(0x7ffd2222e1af, 0x6, 0x7ffd2222e1a7, 0x7, 0xc000226510, 0xc002315c88, 0x0, 0x0)
        /home/circleci/project/builder/build.go:145 +0x628 fp=0xc002315c48 sp=0xc002315650 pc=0xd4c088
main.Build(0x7ffd2222e1af, 0x6, 0x7ffd2222e1a7, 0x7, 0xc0001aa0e0, 0x0, 0x1a)
        /home/circleci/project/main.go:97 +0xc5 fp=0xc002315cb0 sp=0xc002315c48 pc=0xd5c835
main.main()
        /home/circleci/project/main.go:888 +0x1b7f fp=0xc002315f88 sp=0xc002315cb0 pc=0xd6144f
runtime.main()
        /usr/local/go/src/runtime/proc.go:203 +0x1fa fp=0xc002315fe0 sp=0xc002315f88 pc=0x9b84aa
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1373 +0x1 fp=0xc002315fe8 sp=0xc002315fe0 pc=0x9e5381

rax    0x0
rbx    0x7f0075e8e700
rcx    0x7f009d35b18b
rdx    0x0
rdi    0x2
rsi    0x7f0075e8c540
rbp    0x7f009d4d0588
rsp    0x7f0075e8c540
r8     0x0
r9     0x7f0075e8c540
r10    0x8
r11    0x246
r12    0x63bc1c9
r13    0x2642
r14    0x63bed55
r15    0x7f0063ca2920
rip    0x7f009d35b18b
rflags 0x246
cs     0x33
fs     0x0
gs     0x0

I first thought this was an environmental problem on my pc but I was able to build and deploy blinky1 and it works as expected on the device. The GPS I2C example gives me the same error.

Is this an AVR on LLVM issue?

Any help appreciated.

Thank, Tom

deadprogram, 2020-08-20:
Sadly, yes. This GPS driver is really too complex for the current AVR on LLVM. Hopefully when LLVM is released, some of @aykevl code contributions here will help.

tomprouseau, 2020-08-21:
Thanks for getting back to me. I'll put my project on hold until LLVM 11 is released and make another attempt then.

closed 2020-08-21 by tomprouseau