tinygo-org/drivers #188: smoke tests fail

rogpeppe, 2020-08-23:


deadprogram, 2020-08-23:
Looks like you are running version of TinyGo earlier than 0.14.0. Please update. This is likely due to https://github.com/tinygo-org/tinygo/pull/1216

rogpeppe, 2020-08-24:
Sorry, somehow my version paste got erased when editing.
I'm using the tip version of tinygo, at commit 5714559923b9e2aeb133aee8738d4965f51cebfc:
```
% tinygo version
tinygo version 0.15.0-dev linux/amd64 (using go version go1.15 and LLVM version 10.0.1)
```

So I don't _think_ that that's the problem here.

aykevl, 2020-08-24:
Pretty sure you're still using an older version somehow, or an older TinyGo root directory. You could try `tinygo clean`, which might fix it.

sago35, 2020-08-24:
This problem occurs if avr-gcc is not installed.
A PR has been created to have the same response as the tinygo-org/tinygo Makefile.

aykevl, 2020-08-24:
Ah, yes, that last error is indeed fixed by #189. The rest is the result of somehow using an older machine package.

rogpeppe, 2020-08-24:
Where does tinygo look for the machine package? I've run a `make release` and am using the binary from the unpacked `release.tar.gz` file, but perhaps it's looking somewhere else. It does seem like it's finding the machine package somewhere else because I just verified that (for example) `machine.SPI1_SDO_PIN` is defined correctly when tag `pybadge` is set. How does the logic for finding the machine directory work?

rogpeppe, 2020-08-24:
OK, I've found out (kinda) why this issue was happening. Somehow my `release` directory was "corrupted" (possibly by me untarring the release.tar.gz file there at some point in the past), and running `make release` did not fix the issue.

For the record:

```
$ find . -name board_pybadge.go  | xargs md5sum
e5892e49f9f3a481849ff6664105abdf  ./build/release/tinygo/src/machine/board_pybadge.go
7730d24a125d14abbf793ba124975293  ./build/release/tinygo/src/src/machine/board_pybadge.go
7730d24a125d14abbf793ba124975293  ./src/machine/board_pybadge.go
```

The result _should_ have been:
```
7730d24a125d14abbf793ba124975293  ./build/release/tinygo/src/machine/board_pybadge.go
7730d24a125d14abbf793ba124975293  ./src/machine/board_pybadge.go
```

Note the spurious `src/src` directory, which also showed up in the generated release.tar.gz file.
Weirdly, the 7730d24 version there is the correct version.
Perhaps there's a `cp -r` command in a script somewhere which isn't removing the target first?

I'll raise a new issue to track that problem and close this one.

closed 2020-08-24 by rogpeppe

aykevl, 2020-08-25:
When something is messed up in the release directory, it's best to just run `make clean` to start over (which simply removes the build directory). From your previous comments, it wasn't clear to me what your setup was exactly.
Normally I don't build a release and extract that, instead I just copy over the `build/tinygo` file to somewhere in my `$PATH` and run that (for me, that's `~/bin`). It will remember the location it was built and deduce the source directory from that, making incremental changes easy.

rogpeppe, 2020-08-25:
> When something is messed up in the release directory, it's best to just run make clean to start over (which simply removes the build directory).

I think that's a problem - I'd expect the build process to work regardless of the current state.

PR #189 (2020-08-24): Fix smoke-test unless avr-gcc installed
This PR resolves #188 

If I'm going to do it the same way as tinygo-org/tinygo, I think I need to be able to add AVR=0.
I think it is reasonable to add the option because not everyone has avr-gcc installed.
Since the testing of ws2812 is done with circuitplay-express, I think this fix is reasonable.

https://github.com/tinygo-org/tinygo/blob/release/Makefile#L318

before: without avr-gcc installed

```
$ make smoke-test 
...
...
tinygo build -size short -o ./build/test.hex -target=digispark ./examples/ws2812
error: failed to build src/runtime/scheduler_ [... cut]