tinygo-org/drivers #190: release directory isn't properly overwritten by "make release"

rogpeppe, 2020-08-25:
As remarked on by @aykevl [here](https://github.com/tinygo-org/drivers/issues/188#issuecomment-679979139), when the release directory is messed up, one needs to run `make clean` to make it OK again. This is a problem waiting for newcomers to trip over - it would be nice if `make release` could produce a valid release directory regardless of the current contents of that directory.

deadprogram, 2020-08-25:
@rogpeppe I do not think you meant to enter this issue in this repo, did you? Seems like it perhaps should belong in the main repo.

rogpeppe, 2020-08-26:
@deadprogram oops, you're absolutely right, sorry! I just created https://github.com/tinygo-org/tinygo/issues/1321

closed 2020-08-26 by rogpeppe