tinygo-org/drivers #195: Using Rotary Encoders with Blue Pill STM32F103

olablt, 2020-08-30:
I have Blue Pill board and Rotary Encoder connected. I can not find any info on how to implement interrupts on GPIO input pins.

Are there any attempts to make a driver for encoder?

Arduino library code: https://github.com/PaulStoffregen/Encoder/blob/master/Encoder.h

alankrantas, 2020-09-01:
Well...this might work, except that ```machine.PinInputPullup``` has not been implemented on Bluepill yet (I only tested it on a BBC micro:bit). You would have to use external pull-ups.

```golang
package main

import (
	"machine"
	"time"
)

func main() {

	clk := machine.PB9
	dta := machine.PB8
	sw := machine.PB7
	clk.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
	dta.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
	sw.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
	var clkNow, clkPrv bool

	for {

		if !sw.Get() {
			println("Switch pressed")
			time.Sleep(time.Millisecond * 250)
		}

		clkNow = clk.Get()
		if clkNow != clkPrv && clkNow {
			if dta.Get() != clkNow {
				println("Turing clockwise")
			} else {
				println("Turing anti-clockwise")
			}
		}
		clkPrv = clkNow

		time.Sleep(time.Millisecond * 1)

	}

}
```

pavelanni, 2023-07-16:
This code worked for me on RPi Pico (with minor changes in pin names). 
`tinygo version 0.27.0 linux/amd64`
Thanks, @alankrantas !

deadprogram, 2024-12-17:
Closing since the rotary encoder appears to work as expected. Please open an issue on main TinyGo repo if needed for STM32.

Thanks everyone!

closed 2024-12-17 by deadprogram