tinygo-org/drivers #203: MAX6675 river

jhoydich, 2020-10-05:


jhoydich, 2020-10-07:
It seems to be an issue with floats in general. If I declare a float and try to print it with println then the program will compile and upload but won't display to a serial port. When I try to print a float using fmt.Println or Print I have issues compiling. I have also tried converting from a uint16 to a string and then to a float using the strconv library. Converting to a string is no issue. But again an issue during compiling occurs when converting from a string to a float using ParseFloat.

conejoninja, 2020-10-14:
While floats are generally supported, they tend to generate a number of issues and it's better to avoid them when possible. Sensors in tinygo usually works with millis unit (millidegrees instead of degrees) and uint32 to avoid those issues, for example : https://github.com/tinygo-org/drivers/blob/release/bmp280/bmp280.go#L135

Instead of 
```go
t := float64(temp) * .25 // returns degrees
```

```go
t := uint32(temp) * 250 // returns millidegrees
```

Also, ReadCelsius() uint16, should be ReadTemperature() (uint32, error), so you could exchange one sensor for another without refactoring the code

jhoydich, 2020-10-14:
Awesome! Thank you! I'll make those changes soon and test when I am home in a couple weeks. 

I'm guessing I should also change the receiver of ReadTemperature() (as well as Configure() and spiRead()) to be a pointer instead of a value? 

I'm also assuming I should really be passing an spi bus to the New() function and using the built in tx() method to read from the sensor? 

Sorry for all of the questions, this is the first time I've tried writing a driver. I find this project very interesting overall and I hope I can eventually contribute what (little) I can in the future.

conejoninja, 2020-10-15:
If you are using a spi transport you should be passing it already configured to the device New function. 
Existing drivers are a good template for new ones. Don't worry and make as many questions as needed, we're happy to welcome you to our little community and grateful you want to contribute!