tinygo-org/drivers #204: ws2812 driver and interrupts

ewwaller, 2020-10-09:
The ws2812 on an ARM Cortex M4 [Adafruit Feather M4] seems to require that interrupts be disabled during the write.  With the guard, the data sent to the LED controller is occasionally garbled.   I disabled interrupts just before the in line assembly WriteByte in ws2812_m4_120m.go, and re-enabled them immediately after.  The LEDs no longer displayed intermittent flashes (visible when the LED is continuously updated at 500Hz).  As a sanity check, I ensured I was able to accurately catch 10hKz pulse trains on an edge triggered interrupt with the LED updating with Interrupt disable without dropping any interrupts.

Also, the timing is a little wrong -- to hit the pulsewidths in the sec sheet, I had to remove a few NOPs from the initial high period (8, IIRC)

deadprogram, 2020-10-13:
That sounds very interesting. Are you going to submit a PR with those changes, possibly?

ewwaller, 2020-10-14:
I will be happy to.  I'm a little leary as I am still a newcomer to both Go and Tinygo; but I've much experience with embedded processors.  Let me recheck the timing this weekend, and will create a pull request.

aykevl, 2020-10-14:
Previously this was difficult, but now we have the runtime/interrupt package that you can use for this purpose.

deadprogram, 2021-07-02:
This has been released in 0.17.1. Thanks!

closed 2021-07-02 by deadprogram