tinygo-org/drivers #205: DS1307 real time clock can't compile .hex

yevsia, 2020-10-11:
TinyGo version 0.14.1
Go version 1.15

```
package main

import (
	"machine"
	"strconv"
	"time"

	"tinygo.org/x/drivers/ds1307"
)

var (
	uart = machine.UART0
	tx   = machine.UART_TX_PIN
	rx   = machine.UART_RX_PIN
)

func main() {
	machine.I2C0.Configure(machine.I2CConfig{})
	rtc := ds1307.New(machine.I2C0)
	rtc.SetTime(time.Date(2019, 5, 15, 20, 34, 12, 0, time.UTC))

	for {
		t, err := rtc.Time()
		if err != nil {
			println("Error reading date:", err)
			break
		}

		uart.Write([]byte(strconv.Itoa(t.Second())))

	}

}
```

```
Exception 0xc0000005 0x0 0x181 0x2d884d7
PC=0x2d884d7

runtime: unknown pc 0x2d884d7
stack: frame={sp:0x2d55dca0, fp:0x0} stack=[0x0,0x2d55feb0)
000000002d55dba0:  0000000000000000  0000000000000000
000000002d55dbb0:  000005c0fffffb30  000004d0fffffb30
000000002d55dbc0:  00000000000000f0  0000000002d77ea6
000000002d55dbd0:  00000000c0000005  0000000000000000
000000002d55dbe0:  0000000002d884d7  0000000000000002
000000002d55dbf0:  0000000000000000  0000000000000181
000000002d55dc00:  0000000000000000  0000000000000000
000000002d55dc10:  0000000000000000  0000000000000000
000000002d55dc20:  0000000000000000  0000000000000000
000000002d55dc30:  0000000000000000  0000000000000000
000000002d55dc40:  0000000000000000  0000000000000000
000000002d55dc50:  0000000000000000  0000000000000000
000000002d55dc60:  0000000000000000  0000000000000000
000000002d55dc70:  0000000002d884d7  0000000000000000
000000002d55dc80:  00000000041e6a20  000000002d55dca0
000000002d55dc90:  000000002d550000  000000002d55dd70
000000002d55dca0: <000000002ee8d9a0  0000000000000000
000000002d55dcb0:  000000002f4249d8  0000000000000000
000000002d55dcc0:  000000002d55dd50  0000000000000002
000000002d55dcd0:  0000000000000005  0000000000000000
000000002d55dce0:  00000000064b0b60  000000002d55dd90
000000002d55dcf0:  000000002f4427b0  0000000000000000
000000002d55dd00:  0000000000000005  0000000000000000
000000002d55dd10:  00000000779f1440  0000000000000000
000000002d55dd20:  0000000000000001  0000000000360000
000000002d55dd30:  000000002f4ee328  0000000000000090
000000002d55dd40:  0000000000000004  0000000000000000
000000002d55dd50:  000000002f4427b0  0000000000000000
000000002d55dd60:  000000002f4249d8  0000000000000000
000000002d55dd70:  000000002d55dd80  0000002000000009
000000002d55dd80:  2f08e67000000033  2f4427b000000000
000000002d55dd90:  0000000000000000  000000002f4249d8
runtime: unknown pc 0x2d884d7
stack: frame={sp:0x2d55dca0, fp:0x0} stack=[0x0,0x2d55feb0)
000000002d55dba0:  0000000000000000  0000000000000000
000000002d55dbb0:  000005c0fffffb30  000004d0fffffb30
000000002d55dbc0:  00000000000000f0  0000000002d77ea6
000000002d55dbd0:  00000000c0000005  0000000000000000
000000002d55dbe0:  0000000002d884d7  0000000000000002
000000002d55dbf0:  0000000000000000  0000000000000181
000000002d55dc00:  0000000000000000  0000000000000000
000000002d55dc10:  0000000000000000  0000000000000000
000000002d55dc20:  0000000000000000  0000000000000000
000000002d55dc30:  0000000000000000  0000000000000000
000000002d55dc40:  0000000000000000  0000000000000000
000000002d55dc50:  0000000000000000  0000000000000000
000000002d55dc60:  0000000000000000  0000000000000000
000000002d55dc70:  0000000002d884d7  0000000000000000
000000002d55dc80:  00000000041e6a20  000000002d55dca0
000000002d55dc90:  000000002d550000  000000002d55dd70
000000002d55dca0: <000000002ee8d9a0  0000000000000000
000000002d55dcb0:  000000002f4249d8  0000000000000000
000000002d55dcc0:  000000002d55dd50  0000000000000002
000000002d55dcd0:  0000000000000005  0000000000000000
000000002d55dce0:  00000000064b0b60  000000002d55dd90
000000002d55dcf0:  000000002f4427b0  0000000000000000
000000002d55dd00:  0000000000000005  0000000000000000
000000002d55dd10:  00000000779f1440  0000000000000000
000000002d55dd20:  0000000000000001  0000000000360000
000000002d55dd30:  000000002f4ee328  0000000000000090
000000002d55dd40:  0000000000000004  0000000000000000
000000002d55dd50:  000000002f4427b0  0000000000000000
000000002d55dd60:  000000002f4249d8  0000000000000000
000000002d55dd70:  000000002d55dd80  0000002000000009
000000002d55dd80:  2f08e67000000033  2f4427b000000000
000000002d55dd90:  0000000000000000  000000002f4249d8

goroutine 1 [syscall]:
tinygo.org/x/go-llvm._Cfunc_LLVMTargetMachineEmitToMemoryBuffer(0x65043e0, 0x651fc40, 0x1, 0xc003053e78, 0xc003053e80, 0xc000000000)
        _cgo_gotypes.go:9418 +0x54
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer.func1(0x65043e0, 0x651fc40, 0xc000000001, 0xc003053e78, 0xc003053e80, 0x51a0dd)
        C:/Users/VssAdministrator/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x105
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer(0x65043e0, 0x651fc40, 0x1, 0xc00084e680, 0x39, 0x32)
        C:/Users/VssAdministrator/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x8b
github.com/tinygo-org/tinygo/builder.Build(0xc0000a2160, 0x15, 0xc0000a2120, 0x16, 0xc0001b41e0, 0xc002025c68, 0x0, 0x0)
        D:/a/1/s/builder/build.go:145 +0x659
main.Build(0xc0000a2160, 0x15, 0xc0000a2120, 0x16, 0xc000194000, 0x1, 0xffffffffffffffff)
        D:/a/1/s/main.go:97 +0xcc
main.main()
        D:/a/1/s/main.go:888 +0x1cf9
rax     0x2d55df60
rbx     0x2d55df80
rcx     0x2d55df80
rdi     0x6584b70
rsi     0x2d55e550
rbp     0x869
rsp     0x2d55dca0
r8      0xbf
r9      0x2d55e380
r10     0x2d55e3a0
r11     0x2f5021c8
r12     0x2d55e380
r13     0x2f5cb900
r14     0x2d55df90
r15     0x169
rip     0x2d884d7
rflags  0x10206
cs      0x33
fs      0x53
gs      0x2b
```

deadprogram, 2020-10-18:
What target are you trying to compile for? The drivers in this repo are built by the CI system:

https://app.circleci.com/pipelines/github/tinygo-org/drivers/603/workflows/e2cf361d-6c58-45a9-b952-f27932faa401/jobs/612

```
tinygo build -size short -o ./build/test.hex -target=bluepill ./examples/ds1307/sram/main.go
   code    data     bss |   flash     ram
  11096      36    2212 |   11132    2248
96132885a7b5f2eccd4eaefd7cb58e81  ./build/test.hex
tinygo build -size short -o ./build/test.hex -target=bluepill ./examples/ds1307/time/main.go
   code    data     bss |   flash     ram
  17344      92    2292 |   17436    2384
864a3d8924113651f22cfa3e116d03b0  ./build/test.hex
```

yevsia, 2020-10-23:
Target arduino.

aykevl, 2020-10-25:
This is a bug in the AVR backend. I suspect it's the same bug that also prevents the fmt package from being compiled.

Hopefully it'll get fixed eventually. In the meantime, you could try to avoid importing bigger packages like the strconv package that may not yet compile.