tinygo-org/drivers #212: 4x4 membrane Keypad driver

Nerzal, 2020-11-25:
Hi everyone. 

I wrote a 4x4 membrane Keypad driver. 
It currently uses a [][]string matrix internally and returns a string with the button value if a button was pressed. 

I could also change it to use runes instead of strings. 

It currently works on the following default key-mapping. 
I could also add the possibility to inject custom mappings. 

```
	mapping = [4][4]string{
		{"1", "2", "3", "A"},
		{"4", "5", "6", "B"},
		{"7", "8", "9", "C"},
		{"*", "0", "#", "D"},
	}

```

I don't know if i have the time to adapt the driver to also work with a 3x4 keypad right now. So i'd like to add a 4x4 one first and maybe expand it later on to also work with 3x4 keypads. 

Are there any wishes how the driver should behave exactly? 
If not, i'll just create a PR in the next days :)

Nerzal, 2020-11-27:
Update: 

This is more or less what the driver looks like currently: 

`// Driver is a driver for 4x4 keypads
type Driver struct {
	inputEnabled bool
	lastColumn   int
	lastRow      int
	columns      [4]machine.Pin
	rows         [4]machine.Pin
	mapping      [4][4]string
}

// Configure takes r4 -r1 pins and c4 - c1 pins
func (keypad *Driver) Configure(r4, r3, r2, r1, c4, c3, c2, c1 machine.Pin) {
	inputConfig := machine.PinConfig{Mode: machine.PinInputPullup}
	c4.Configure(inputConfig)
	c3.Configure(inputConfig)
	c2.Configure(inputConfig)
	c1.Configure(inputConfig)

	keypad.columns = [4]machine.Pin{c4, c3, c2, c1}

	outputConfig := machine.PinConfig{Mode: machine.PinOutput}
	r4.Configure(outputConfig)
	r3.Configure(outputConfig)
	r2.Configure(outputConfig)
	r1.Configure(outputConfig)

	r4.High()
	r3.High()
	r2.High()
	r1.High()

	keypad.rows = [4]machine.Pin{r4, r3, r2, r1}

	keypad.mapping = [4][4]string{
		{"1", "2", "3", "A"},
		{"4", "5", "6", "B"},
		{"7", "8", "9", "C"},
		{"*", "0", "#", "D"},
	}

	keypad.inputEnabled = true
	keypad.lastColumn = -1
	keypad.lastRow = -1
}

func (keypad *Driver) GetKey() string {
	row, column := keypad.GetIndices()
	if row == -1 && column == -1 {
		return ""
	}

	return keypad.mapping[row][column]
}

func (keypad *Driver) GetIndices() (int, int) {
	for rowIndex := range keypad.rows {
		rowPin := keypad.rows[rowIndex]
		rowPin.Low()

		for columnIndex := range keypad.columns {
			columnPin := keypad.columns[columnIndex]

			if !columnPin.Get() && keypad.inputEnabled {
				keypad.inputEnabled = false

				keypad.lastColumn = columnIndex
				keypad.lastRow = rowIndex

				return keypad.lastRow, keypad.lastColumn
			}

			if columnPin.Get() &&
				columnIndex == keypad.lastColumn &&
				rowIndex == keypad.lastRow &&
				!keypad.inputEnabled {
				keypad.inputEnabled = true
			}
		}

		rowPin.High()
	}

	return -1, -1
}
`

aykevl, 2020-12-01:
I find it hard to say what the API should look like, especially because there are no other keyboard drivers in the repo. However, I'm a bit unsure about the current layout. Maybe it can be changed to return key codes instead (indicating the position on the pad)? I believe this is how keyboards work internally. Comparing numbers is also much faster/simpler and they can be converted to names easily if necessary:

```go
var keyCodes = [...]string{
    0: "1",
    1: "2",
    2: "3",
    3: "A",
    4: "4",
    5: "5",
    6: "6",
    7: "B",
    // ...etc
}

// ...
keyCode := ... // read keycode from keypad
keyString := keyCodes[keyCode]
```

deadprogram, 2021-05-12:
This has now been released, so closing. Thanks!

closed 2021-05-12 by deadprogram