tinygo-org/drivers #218: VSync with ST7789 display wiring problem

anydream, 2020-12-29:
Hello,

I noticed that the ST7789 driver has implements the VSync function, but my ST7789 display has only one SDA pin for SPI communication.
How to wiring ESP32 and ST7789 display to make the VSync work?

aykevl, 2021-01-06:
I have not personally tested the ST7789 driver on the ESP32 and I haven't attempted reading a register from the display either.

To be honest, I don't know how this should be wired. Maybe it requires some diodes and resistors to split the signal?

anydream, 2021-01-06:
After several days of research, I found that my display pinout is 3-wire SPI: CS/SCK/SDA.
reading and writing can be done through only one SDA pin.
Maybe this issue is solved.

conejoninja, 2021-02-13:
Last commit on release branch ( #ce9c93f ) changed the ST7789 example SPI mode from 3 to 0. @anydream  Is that the cause of your problem? The example stopped working on my 3-wire SPI display until I change back the mode to 3. If that's the case, we should close this issue.

closed 2021-02-22 by anydream