tinygo-org/drivers #221: RAM overflow while trying to use espat

quaintdev, 2021-01-09:
I am trying to communicate with esp using Arduino uno. Below is the simple program I am trying to execute.

`func main() {
	esp := machine.UART{}
	esp.Configure(machine.UARTConfig{
		BaudRate: 115200,
		RX:       machine.UART_RX_PIN,
		TX:       machine.UART_TX_PIN,
	})
	println("Hello World")
	espDevice := espat.New(esp)
	ip, err := espDevice.GetClientIP()
	if err!=nil{
		println(err)
	}
	println(ip)
}
`
The program works fine until espDevice initialization. Any attempt to use espDevice methods results into RAM overflow error as shown below.

`
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: /tmp/tinygo726373768/main section `.data' will not fit in region `RAM'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: region `RAM' overflowed by 3026 bytes
/tmp/tinygo726373768/main.o: In function `LBB9_3':
main.go:(.text.(reflect.Value).Complex+0x9a): undefined reference to `__extendsfdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.(reflect.Value).Complex+0xba): undefined reference to `__extendsfdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found address size '9', this reader cannot handle sizes greater than '8'.
/tmp/tinygo726373768/main.o: In function `LBB11_3':
main.go:(.text.(reflect.Value).Float+0x3c): undefined reference to `__extendsfdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '12', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo726373768/main.o: In function `runtime.printfloat64':
main.go:(.text.runtime.printfloat64+0x46): undefined reference to `__unorddf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x64): undefined reference to `__gtdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x7a): undefined reference to `__adddf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '4354', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0xac): undefined reference to `__nedf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '13312', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0xd6): undefined reference to `__ltdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '256', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0xf8): undefined reference to `__nedf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '65535', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x170): undefined reference to `__nedf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '17', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x1f0): undefined reference to `__divdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '15619', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x200): undefined reference to `__ltdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x228): undefined reference to `__ltdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x25e): undefined reference to `__gedf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x27a): undefined reference to `__divdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '1537', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x2b2): undefined reference to `__muldf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x2ec): undefined reference to `__ltdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x314): undefined reference to `__adddf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '1792', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x32e): undefined reference to `__gedf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x356): undefined reference to `__divdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '18946', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x3ac): undefined reference to `__fixdfsi'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '13312', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x3c4): undefined reference to `__floatsidf'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '256', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x3d8): undefined reference to `__subdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '65535', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.printfloat64+0x3f2): undefined reference to `__muldf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '74', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo726373768/main.o: In function `LBB46_15':
main.go:(.text.runtime.reflectValueEqual+0x16a): undefined reference to `__eqdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '23042', this reader only handles version 2, 3 and 4 information.
main.go:(.text.runtime.reflectValueEqual+0x190): undefined reference to `__eqdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '13312', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo726373768/main.o: In function `LBB46_29':
main.go:(.text.runtime.reflectValueEqual+0x316): undefined reference to `__eqdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '256', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo726373768/main.o: In function `LBB57_225':
main.go:(.progmem.data+0xde8): undefined reference to `__ashrsi3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '65535', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo726373768/main.o: In function `LBB57_234':
main.go:(.progmem.data+0xe84): undefined reference to `__ashrsi3'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo726373768/main: exit status 1
`

Any suggestions?

deadprogram, 2021-01-28:
The capabilities of the Arduino Uno are much more limited than other more powerful MCUs such as the 32-bit ARM processors. This is partly due to memory, and also due to limitation in the LLVM backend for AVR. I have only done WiFi from more capable processors so far as a result.

deadprogram, 2024-12-17:
Closing due to age. Please open a new issue if needed. Thank you!

closed 2024-12-17 by deadprogram