tinygo-org/drivers #223: HC-SR04 reads 0mm no matter what

patrickhener, 2021-01-25:
I followed the example of HC-SR04 and tried the official drivers package.

But no matter what I do it always reads 0mm.

What am I doing wrong and which additional information can I provide?

```bash
tinygo version 0.16.0 linux/amd64 (using go version go1.15.7 and LLVM version 11.0.1)
```

On my arch linux machine
```bash
tinygo build -o arduino.hex -size=short -target=arduino-mega2560 arduino.go
```

On my pi connected to Arduino
```bash
 avrdude -p m2560 -D -c wiring -P /dev/ttyACM0 -b 115200 -v -F -U flash:w:arduino.hex
```

Main func
```go
	sensor := hcsr04.New(machine.D9, machine.D9)
	sensor.Configure()

	println("Ultrasonic starts")
	for {
		println("Distance:", sensor.ReadDistance(), "mm")

		time.Sleep(100 * time.Millisecond)
	}
```

Thanks for the help

conejoninja, 2021-01-25:
Hello patrick, I'm assuming hardware is fine and everything connected as it should. Please check pin numbers and that trigger/echo correct too.

Try sensor.ReadPulse instead of sensor.ReadDistance, if it still returns 0, then the timeout is reached, which means the echo pin didn't get a signal.

patrickhener, 2021-01-26:
I don't think it is timeout to be honest. I added this code parts at the timeout statement:

```go
		if i > 10 {
			if time.Since(t).Microseconds() > TIMEOUT {
				println("Timeout reached")
				return 0
			}
			i = 0
		}
```

And the result is not `Timeout reached` before I get the `0 mm` output.

I double- and triple-checked pins which are correct. Do I need to somehow init() the PWM mode on my Arduino Mega2560 with tinygo as I would need for ADC?

Just for reference here is a ino-sketch which works:

```C
const int echoPin = 8;
const int trigPin = 9;
long duration;
int distance;

void setup() {
	pinMode(trigPin, OUTPUT);
	pinMode(echoPin, INPUT);
	Serial.begin(9600);
}

void loop() {
  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  duration = pulseIn(echoPin, HIGH);
  distance = duration*0.034/2;
  Serial.println(distance);
  delay(100);
}
```

The output of it will look like:
```
4
4
4
4
26
26
27
26
26
...
```

So pins and hardware is fine. Any suggestions?

conejoninja, 2021-01-26:
There are 3 parts where it returns 0, ln 62, 75 and 80. Did you put the println in all of them?

patrickhener, 2021-01-26:
This is my hcsr04.go
```go
// Package hcsr04 provides a driver for the HC-SR04 ultrasonic distance sensor
//
// Datasheet:
// https://cdn.sparkfun.com/datasheets/Sensors/Proximity/HCSR04.pdf
package hcsr04

import (
	"machine"
	"time"
)

const TIMEOUT = 23324 // max sensing distance (4m)

// Device holds the pins
type Device struct {
	trigger machine.Pin
	echo    machine.Pin
}

// New returns a new ultrasonic driver given 2 pins
func New(trigger, echo machine.Pin) Device {
	return Device{
		trigger: trigger,
		echo:    echo,
	}
}

// Configure configures the pins of the Device
func (d *Device) Configure() {
	d.trigger.Configure(machine.PinConfig{Mode: machine.PinOutput})
	d.echo.Configure(machine.PinConfig{Mode: machine.PinInput})
}

// ReadDistance returns the distance of the object in mm
func (d *Device) ReadDistance() int32 {
	pulse := d.ReadPulse()

	// sound speed is 343000 mm/s
	// pulse is roundtrip measured in microseconds
	// distance = velocity * time
	// 2 * distance = 343000 * (pulse/1000000)
	return (pulse * 1715) / 10000 //mm
}

// ReadPulse returns the time of the pulse (roundtrip) in microseconds
func (d *Device) ReadPulse() int32 {
	t := time.Now()
	d.trigger.Low()
	time.Sleep(2 * time.Microsecond)
	d.trigger.High()
	time.Sleep(10 * time.Microsecond)
	d.trigger.Low()
	i := uint8(0)
	for {
		if d.echo.Get() {
			t = time.Now()
			break
		}
		i++
		if i > 10 {
			if time.Since(t).Microseconds() > TIMEOUT {
				println("Timeout reached")
				return 0
			}
			i = 0
		}
	}
	i = 0
	for {
		if !d.echo.Get() {
			return int32(time.Since(t).Microseconds())
		}
		i++
		if i > 10 {
			if time.Since(t).Microseconds() > TIMEOUT {
				println("Timeout reached")
				return 0
			}
			i = 0
		}
	}
	println("Timeout reached")
	return 0
}
```

This is my main.go
```go
package main

import (
	"machine"
	"time"

	hcsr04 "github.com/dascr/dascr-machine/arduino/hcsrc04"
)

const (
	// Ultrasonic
	echoPin    = machine.D8
	triggerPin = machine.D9
)

func main() {
	ultrasonic := hcsr04.New(echoPin, triggerPin)
	ultrasonic.Configure()

	for {
		distance := ultrasonic.ReadDistance()
		println(distance)
		time.Sleep(time.Millisecond * 100)
	}
}
```

conejoninja, 2021-01-26:
I'll need to find my atmega2560 and try to reproduce your issue, I'm out of ideas since ReadPulse should not be returning 0 if it's not a timeout/invalid measure

patrickhener, 2021-01-26:
okay thanks for the effort. I really appreciate it. I am keen to see if you get it to work then. Thanks

Nerzal, 2021-01-31:
Could be just related to the AVR backend
[image]

patrickhener, 2021-02-01:
So I guess I need to stick to C++/Arduino IDE for now.

conejoninja, 2021-02-01:
I was afraid it was related, that's why I wanted to check myself. I will close this issue then.

closed 2021-02-01 by conejoninja