tinygo-org/drivers #233: WIFININA: WLAN connection cannot be established

Nerzal, 2021-02-25:
I had some problems maintaining a stable connection with Firmware version 1.2.1, so i tried nearly every other Firmware version i could find. 

The problem that i do now face is, that I'm unable to connect to the WLAN now. Also tried it with the connectToAP() function from the examples, which also does not lead to an established connection anymore. 

To verify, that i did not break anything when flashing a new firmware, i tried the Arduino IDE example sketch, which successfully connects and maintains a connection. 

So does anyone have a trick, what to do?

Example function:

```go
func connectToAP() {
	time.Sleep(2 * time.Second)
	println("Connecting to " + ssid)
	adaptor.SetPassphrase("mySSID", "mySecret")
	for st, _ := adaptor.GetConnectionStatus(); st != wifinina.StatusConnected; {
		println("Connection status: " + st.String())
		time.Sleep(1 * time.Second)
		st, _ = adaptor.GetConnectionStatus()
	}
	println("Connected.")
	time.Sleep(2 * time.Second)
	ip, _, _, err := adaptor.GetIP()
	for ; err != nil; ip, _, _, err = adaptor.GetIP() {
		println(err.Error())
		time.Sleep(1 * time.Second)
	}
	println(ip.String())
}
```

Output: 

```
Connection status: No SSID Available
Connection status: No SSID Available
Connection status: No SSID Available
Connection status: No SSID Available
Connection status: Disconnected
Connection status: Disconnected
Connection status: Disconnected
....
```

Nerzal, 2021-02-26:
I'm currently trying to analyze whats going on here. 

My router logs arent that helpful, but atleast i can see, that the arduino actually talks to the router. 

```
(W005) WLAN-Anmeldung via <2.4> Ghz Frequenzband ist gescheitert: Autorisierung fehlgeschlagen. Rechnername: <arduino-nano-iot-33>, Mac-Adresse: <24:62:AB:B8:F8:84>
``` 

That roughly translates to: Connection attempt to 2.4 Ghz failed: Authorization failed. 
So i think, that maybe the password isn't being transmitted correctly.

Nerzal, 2021-02-26:
For documentation purposes so i (and others) can find the interesting changes in the wifinina driver lib for newer firmware versions.

Nerzal, 2021-05-11:
Closing this issue, as it is fixed :)

deadprogram, 2021-05-11:
Reopening, will close once released. Has been labeled for "next release".

ysoldak, 2021-05-12:
Close already? 😸

closed 2021-05-12 by deadprogram