tinygo-org/drivers #238: Support ENC28J60 ethernet card

soypat, 2021-03-13:
[the IC](https://www.microchip.com/wwwproducts/en/en022889).

If no one is working on implementing it, I'll do it! I've already done work to port https://github.com/muanis/arduino-projects, specifically [this file](https://github.com/muanis/arduino-projects/blob/master/libraries/etherShield/enc28j60.c).

I'm just a little unsure how testing works with TinyGo.

aykevl, 2021-03-13:
> I'm just a little unsure how testing works with TinyGo.

We do have some tests, but nothing big yet. I'd focus first on getting it working in the first place.

Georges760, 2021-04-09:
I did a port on STM32 with STM32CubeF4 : https://github.com/Georges760/STM32_ENC28J60
Maybe it can help.

soypat, 2021-04-10:
I'll comment on the progress: I've managed to port parts of the arduino ENC28j60 library but I am having trouble reading/writing anything at all through the SPI connection. I think I may have fried my enc28j60. I'm still working on it

Link to progress https://github.com/soypat/drivers/tree/dev/enc28j60

soypat, 2021-04-13:
I honestly can't get it to work for the life of me. I've created tests to read/write to registers and they are just not being written. If someone with more knowledge on the inner workings of SPI on tinygo would like to try it out... 

[https://github.com/soypat/drivers/tree/dev/enc28j60](https://github.com/soypat/drivers/tree/dev/enc28j60)