tinygo-org/drivers #245: nRF24L01 2.4GHz Transceiver

mortiy, 2021-04-03:
Hi,
Is there any progress on pretty common 2.4 GHz transceiver `nRF24`?

Datasheet: 
https://www.mouser.com/datasheet/2/297/nRF24L01_Product_Specification_v2_0-9199.pdf

efectn, 2021-09-22:
I'll try to add driver for nrf24l01 in next weeks

mytechnotalent, 2022-03-26:
@efectn have you made a TinyGo nRF24 driver and example?  I am trying to use this on the Raspberry Pi Pico.

efectn, 2022-03-26:
> @efectn have you made a TinyGo nRF24 driver and example? I am trying to use this on the Raspberry Pi Pico.

Hi. I didn't have much time to do it. Sorry :/

mytechnotalent, 2022-03-27:
@efectn here is a video showing how to build this driver in MicroPython with a nRF24L01 and a Raspberry Pi Pico.  I would implement this myself but I am not strong on driver development.  Getting the nRF24L01 working with TinyGo would really open up the possibilities for TinyGo outside of Bluetooth for communication.
https://www.youtube.com/watch?v=lnE4m6SqZ5E&t=82s 
https://www.youtube.com/watch?v=6-C9xnlXKTg&t=615s

asterisk800, 2022-05-21:
While searching for a diver for NRF24L01, I came across this post. Any progress on this?

StarkovOleksandr, 2022-08-09:
Hi. Can I offer add to drivers-repo my version of driver for NRF24L01?

deadprogram, 2022-08-09:
@StarkovOleksandr that would be fantastic! Yes, please.

StarkovOleksandr, 2022-08-10:
@deadprogram 
What flow you use?
Have I to make new branch "nRF24L01", and load it there?

deadprogram, 2022-08-10:
@StarkovOleksandr please see https://github.com/tinygo-org/drivers/blob/release/CONTRIBUTING.md#some-specific-hardware-you-want-to-use-does-not-appear-to-be-in-the-tinygo-drivers

StarkovOleksandr, 2022-08-11:
@deadprogram, in the drivers repo no tests for doing step "Make sure the tests still pass". Can I miss this step?

deadprogram, 2022-08-11:
https://github.com/tinygo-org/drivers/blob/release/CONTRIBUTING.md#how-to-run-tests 

however perhaps we should be explicit about doing that in the previous section?

StarkovOleksandr, 2022-08-11:
@deadprogram, no-no. It's wrote clearly.  I seen it. 
But got mistake 
```
process_begin: CreateProcess(NULL, rm -rf build, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:3: clean] Error 2
```
Maybe I need to fix some thing in my environments.
Okay, I'll deal with this.
Thanks.

StarkovOleksandr, 2022-08-11:
It seems like I fully stuck (  Sorry.
I don't know what is mean this error.
Which changes I didn't? Should I supplement `Makefile` and `CHANGELOG.md`?
Or this error relate to something else?

tapir, 2022-12-28:
any update on the merge?

StarkovOleksandr, 2023-01-01:
@tapir, hi.
If you need driver just now, you can just add it to your project from here https://github.com/StarkovOleksandr/drivers/tree/dev/nrf24l01

For end the merge I need a bit learn about mistakes and maybe about processes of merging. 
But I'm from Ukraine and orcs changed of my priorities. I can't give any terms when I will close this ticket.
But if you can help closed the ticket we can connect directly and I finally end it :)