tinygo-org/drivers #255: DHTXX driver does not work on Arduino Uno

adrianbn, 2021-04-30:
When trying to build the example for the dht driver on Windows I get the following error:

```# tinygo.org/x/drivers/dht
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.15.1\dht\thermometer.go:160:10: UART1 not declared by package machine
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.15.1\dht\thermometer.go:161:16: UART1 not declared by package machine
```

Taking a quick look at the code for the driver, it tries to access UART1 https://github.com/tinygo-org/drivers/blob/release/dht/thermometer.go#L160, which according to the `machine` package for arduino is not defined (https://tinygo.org/microcontrollers/machine/arduino/).

I'm not sure if the fix is as simple as renaming the constant to UART instead, and whether that'd break other boards.

Cheers!

sago35, 2021-04-30:
It's not a good idea to use machine.UART1 in dht/thermometer.go.
It is better to pass it from dht.New(), for example, so that it does not depend on a specific UART.

adrianbn, 2021-04-30:
Not sure I follow. Do you mean pass the UART from the device into `dht.New()` and then trickle it down to `receiveSignals`? Even then, it doesn't seem like the arduino UART in the machine package supports `Interrupt()`, right?

mikespook, 2021-05-30:
To the boards without [UART](https://github.com/tinygo-org/drivers/blob/release/dht/thermometer.go#L160), it causes a problem to build. This is issue.

aykevl, 2021-06-01:
The correct fix would be to use the `runtime/interrupt` package:

```go
mask := interrupt.Disable()
// do something with interrupts disabled
interrupt.Restore(mask)
```

That seems reasonable for the DHTXX driver as it depends on precise timings.