tinygo-org/drivers #276: HTS221 support

deadprogram, 2021-06-20:
We need support for the HTS221 capacitive digital sensor for relative humidity and temperature.

It is one of the sensors onboard the Arduino Nano 33 BLE Sense.

deadprogram, 2021-06-20:
See https://content.arduino.cc/assets/Nano_BLE_Sense_HTS221.pdf

sago35, 2021-06-23:
The following may be helpful.

https://github.com/arduino-libraries/Arduino_HTS221

alankrantas, 2021-08-22:
Tried to look into it. I tried to raise a response with the "who am I" register but... the code would simply hang on any I2C operations.

```golang
func main() {

	const HTS221_ADDRESS = 0x5F
	const HTS221_WHO_AM_I_REG = 0x0F

	i2c := machine.I2C1
	i2c.Configure(machine.I2CConfig{
		SCL:       machine.P0_15,  // SCL1
		SDA:       machine.P0_14,  // SDA1
	})

	data := []byte{0}
	err := i2c.ReadRegister(HTS221_ADDRESS, HTS221_WHO_AM_I_REG, data)
	println(data[0], err)

}
```

[schematics](https://content.arduino.cc/assets/NANO33BLE_V2.0_sch.pdf)

If I set to other pins or set nothing I would get an "I2C bus error". I couldn't get "who am I" response from LSM9DS1 either (also stuck). I can ping all onboard devices with the Arduino I2C scanner script.

Is there any way to check if the SCL1/SDA1 pins are working properly in TinyGo?

sago35, 2021-08-23:
@alankrantas 
I tried it with feather-nrf52840-sense, which has the same nRF52840 microcontroller.
It did not work when the I2C initialization part was set to `I2C1`.

Try changing it to `I2C0` and see if it works.
If it does, there may be some changes needed in `machine_nrf.go`.

```go
	// It works (I2C0)
	machine.I2C0.Configure(machine.I2CConfig{
		Frequency: machine.TWI_FREQ_400KHZ,
		SDA:       machine.P0_14,
		SCL:       machine.P0_15,
	})
```

```go
	// It did not work (I2C1)
	machine.I2C1.Configure(machine.I2CConfig{
		Frequency: machine.TWI_FREQ_400KHZ,
		SDA:       machine.P0_14,
		SCL:       machine.P0_15,
	})
```

sago35, 2021-08-23:
NewI2C(). I forgot to change the part that passes to ssd1306.
I2C0 and machine.I2C1 both worked.
The above comment has been corrected.

sago35, 2021-08-23:
@alankrantas 
You probably need to set VDD_ENV (P0_22).
This is an output pin and probably needs to be set high to turn on the HTS221 power supply.

[image]

alankrantas, 2021-08-23:
@sago35 Thanks! I got responses from both HTS221 and LSM9DS1.

It turns out I have to set both ```VCC_ENV (P0_22)``` (device power)  and ```R_PULLUP (P1_00)``` (pull up the I2C1 bus) to high and **wait a bit**, like 5 ms at least.