tinygo-org/drivers #282: RTL8720DN: ConnectToAP() does not proceed as expected

deadprogram, 2021-07-02:
When using the examples for the RTL8720DN and the `debug=true`, the program does call `ConnectToAP()` until the serial port from the computer connects to the board, and then does not proceed with the `ConnectToAP()` call until the user disconnects the serial connection. This is very odd behavior.

sago35, 2021-07-03:
WaitSerial() is executed only when debug=true, but this is undocumented and confusing to the user.
I will figure out how to fix it.
I think it would be better to put it in the documentation first.

https://github.com/tinygo-org/drivers/blob/release/examples/rtl8720dn/webclient-tinyterm/wioterminal.go#L54-L56

sago35, 2021-07-03:
https://www.twitch.tv/videos/1074459848?t=01h09m58s

In your video, you were unable to proceed from rpc_wifi_connect().
In my environment, if I wait 5 seconds, I get a response (rx).
In your video, you terminated picocom, but what happens if you don't terminate it?

deadprogram, 2022-01-22:
This was already addressed I think, so now closing.

closed 2022-01-22 by deadprogram