tinygo-org/drivers #290: examples/wifinina/tlsclient failed

sago35, 2021-08-05:
If the URL is changed to the following, no error will be displayed.

* www.google.com
* github.com

The environment is as follows.

* windows10
* pyportal + NINA_W102-1.7.4.bin
* tinygo-org/drivers dev (9234d7d9ffee39a7eb83239a90aed15f288655cf)
* tinygo e834d7887143fc8da542d3d29907e8db7a5b9bb2 (2021/07/16)

```
$ tinygo flash --target pyportal --size short --programmer jlink ./examples/wifinina/tlsclient/
```

```
Connecting to XXXXXX
Connection status: No SSID Available
Connection status: No SSID Available
Connection status: No SSID Available
Connection status: No SSID Available
Connection status: No SSID Available
Connected.
192.168.1.123

---------------
Dialing TCP connection
Connection failed: wifinina error: 0x01
```

scottfeldman, 2023-01-31:
Just a note for now: this is fixed in netdev

deadprogram, 2024-12-17:
Closing since no longer valid. Thanks everyone!

closed 2024-12-17 by deadprogram