tinygo-org/drivers #294: How about adding ConnectToAP to net.DeviceDriver interface?

sago35, 2021-08-23:
If we want to use tinygo-org/drivers/net, we must connect to AccessPoint.
I have implemented `ConnectToAP()` on rtl8720dn, but not on espat and wifinina.
By implementing ConnectToAP(), espat/wifinina/rtl8720dn will be able to use almost the same code.

`net.DeviceDriver` interface
https://github.com/tinygo-org/drivers/blob/a0119ffc6f09c088e1bf93639e8775fd8747d7e4/net/driver.go#L3-L17

`rtl8720dn.ConnectToAP(ssid string, password string) error`
https://github.com/tinygo-org/drivers/blob/a0119ffc6f09c088e1bf93639e8775fd8747d7e4/rtl8720dn/wifi.go#L8

ref: https://github.com/tinygo-org/drivers/pull/273#discussion_r659211641

deadprogram, 2022-01-17:
@sago35 please see https://github.com/tinygo-org/drivers/pull/357 where I am working on something related to this.

deadprogram, 2022-09-28:
I think this is done now @sago35

sago35, 2022-09-28:
Yes, this Issue can be closed.

https://github.com/tinygo-org/drivers/blob/1bb1b621c6cfa60544ff89c510da46858dbe29bb/net/adapter.go#L16-L18

closed 2022-09-29 by deadprogram