tinygo-org/drivers #301: Multiple Drivers fail to compile on Raspberry Pico

SoftTacos, 2021-08-31:
Hi there, two things before I get to my post. First, I'm new to the realm of microcontrollers, so I could be(probably am) missing something here. Second, I just want to express gratitude on the existence of TinyGo, is fantastic and usually easier to just build things myself on my pico in TinyGo.
I'm running tinygo from my windows10 machine with drivers v0.17.1, tinygo v0.19.0 and have been able to get code that I wrote from scratch to work just fine. I'm guessing these errors are because something isn't supported yet for the pico, however I don't know what I don't know here and want to make sure I'm not doing anything pants-on-head.

The first compile error I got is with the [DHT drivers](https://github.com/tinygo-org/drivers/blob/release/examples/dht/main.go):
```
d := dht.New(machine.GP14, dht.DHT22)
d.Measurements()
```
I ran: `tinygo flash -target=pico ./`
And got:
```
# tinygo.org/x/drivers/dht
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\constants.go:105:10: undeclared name: counter
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\constants.go:112:29: undeclared name: counter
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\thermometer.go:158:47: undeclared name: counter
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\thermometer.go:170:40: undeclared name: counter
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\util.go:18:49: undeclared name: counter
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\constants.go:113:18: CPUFrequency not declared by package machine
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\constants.go:115:9: undeclared name: counter
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\thermometer.go:127:21: undeclared name: counter
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\dht\util.go:19:9: undeclared name: counter
```
I wrote my own drivers on my raspberry pi, ported that over to my pico, and those work great. If there is some hardware limitation preventing the pico from using the standard drivers, let me know and I'll happily get a PR going to add mine to the codebase! I just wanted to bring this up because I wasn't sure if it was a problem on my end or a "bug"/unintentionally missing feature.

Problem 2: When I try to use [ws2812 drivers](https://github.com/tinygo-org/drivers/tree/release/ws2812) with simple test code:
```
pixelpin := machine.GP6
neopixel := ws2812.New(pixelpin)
```
I get: 
```
# tinygo.org/x/drivers/ws2812
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\ws2812\ws2812.go:26:5: d.WriteByte undefined (type Device has no field or method WriteByte)
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\ws2812\ws2812.go:35:5: d.WriteByte undefined (type Device has no field or method WriteByte)
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\ws2812\ws2812.go:36:5: d.WriteByte undefined (type Device has no field or method WriteByte)
..\..\go\pkg\mod\tinygo.org\x\drivers@v0.17.1\ws2812\ws2812.go:37:5: d.WriteByte undefined (type Device has no field or method WriteByte)
```

sago35, 2021-08-31:
> The first compile error I got is with the DHT drivers:

I was able to reproduce it.
There seem to be two problems.

The first is that the counter definition is failing because of the build tag.
This is a problem with the way highfreq.go and lowfreq.go are built.

The other is that the CPUFrequency of the RP2040 is not defined in the machine package.
I think it needs to be added.

> Problem 2: When I try to use ws2812 drivers with simple test code:

I was able to reproduce it.

The driver for ws2812 is target specific and does not support rp2040 at this time.
The following PR may be of help.

https://github.com/tinygo-org/drivers/pull/292
https://github.com/tinygo-org/drivers/pull/217

SoftTacos, 2021-09-01:
Got it, that would explain the DHT and frequency problems.

I also noticed that PortMaskSet() and PortMaskClear() are not defined in machine for RP2040. Is that also being worked on?

sago35, 2021-09-01:
> I also noticed that PortMaskSet() and PortMaskClear() are not defined in machine for RP2040. Is that also being worked on?

Maybe no one is working on it.
It would be nice if you could create https://github.com/tinygo-org/tinygo/pulls .

SoftTacos, 2021-09-02:
I sincerely would if I knew how to do it. I've looked at the other implementations and don't understand what's happening there. If you know where I can find that information I would love to try.

sago35, 2021-09-13:
@SoftTacos 
The value returned by `PortMaskSet()` will probably be the one used in `Pin.set()` below.

https://github.com/tinygo-org/tinygo/blob/64d048c47c95bc0a70b4cff109afc1f9039705a4/src/machine/machine_rp2040_gpio.go#L73-L83
ref: https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf

I found the same thing in the AVR source.

https://github.com/tinygo-org/tinygo/blob/64d048c47c95bc0a70b4cff109afc1f9039705a4/src/machine/machine_avr.go#L76-L85

jeremyforan, 2021-11-04:
Is there a short term fix I could apply to get this working?  I am running into the same errors trying to use the ws2812 with the RP2040.