tinygo-org/drivers #312: machine.I2C to drivers.I2C

trevatk, 2021-09-14:
This is probably a dumb question but I'm very new to Go and embedded development. I saw there was a change to I2C and the examples are no longer the correct way to initialize a sensor. Could I get some help with the proper way to initialize a sensor using I2C? 

```golang
	machine.I2C0.Configure(machine.I2CConfig{})
	sensor := bme280.New(machine.I2C0)
	sensor.Configure()
```
I was trying to read the docs and it seems I need to create a new drivers.I2C with the SDA and SCL number on the machine of the sensor? 

Sorry if I'm not saying any of this correctly.

aykevl, 2021-09-15:
> I was trying to read the docs and it seems I need to create a new drivers.I2C

Interfaces in Go work a bit differently than in other languages. In Go, you can have a named type (a struct or any other type) that satisfies a given interface. So instead of a concrete type explicitly implementing an interface, the concrete type automatically implements any interface that matches it.

In this case, you simply need to create the right `machine.I2C` object and it will be accepted by `bme280.New` because it will satisfy the interface.

> I saw there was a change to I2C and the examples are no longer the correct way to initialize a sensor.

Not? I think the examples are still correct.

trevatk, 2021-09-21:
@aykevl Thank you for help with the interfaces! 

I'm not entirely sure what I'm doing wrong at this point, the wiring is giving power from the esp32-pico-d4 to the adxl345. However, a default 0 value is all that is being returned. 

Both ReadAcceleration and ReadRawAcceleration is throwing a 0 for all x, y, and z values....

aykevl, 2021-09-21:
Can you share the hardware and the code that you're using?

trevatk, 2021-09-21:
Hardware: ESP32-PICO-D4
Sensor: ADXL345 / BME280

main.go
```golang
package main

import (
	"fmt"
	"strconv"
	"time"
	"tiny/pkg/axis"
	"tiny/pkg/temp"

	"tinygo.org/x/drivers/adxl345"
	"tinygo.org/x/drivers/bme280"
)

func main() {

	axis := &axis.Axis{}
	sensor := adxl345.New(axis)

	sensor.Configure()

	temp := &temp.Temp{}
	s := bme280.New(temp)

	s.Configure()

	connected := s.Connected()
	if !connected {
		fmt.Println("BME280 not detected")
		return
	}

	for {
		x, y, z, _ := sensor.ReadAcceleration()
		fmt.Println("X:", x, "Y:", y, "Z:", z)

		rx, ry, rz := sensor.ReadRawAcceleration()
		fmt.Println("X (raw):", rx, "Y (raw):", ry, "Z (raw):", rz)

		temp, _ := s.ReadTemperature()
		println("Temperature:", strconv.FormatFloat(float64(temp)/1000, 'f', 2, 64), "C")

		time.Sleep(2000 * time.Millisecond)
	}
}
```

adxl345.go
```golang
package axis

type Axis struct{}

func (adxl *Axis) ReadRegister(addr uint8, r uint8, buf []byte) error {
	return nil
}

func (adxl *Axis) WriteRegister(addr uint8, r uint8, buf []byte) error {
	return nil
}

func (adxl *Axis) Tx(addr uint16, w, r []byte) error {
	return nil
}
```

sago35, 2021-09-22:
@trevatk 

No i2c support for esp32 has been created yet.
In #319 I created software I2C. If you use the i2csoft branch, you can write it as follows.

```go
package main

import (
	"fmt"
	"machine"
	"strconv"
	"time"

	"tinygo.org/x/drivers/adxl345"
	"tinygo.org/x/drivers/bme280"
	"tinygo.org/x/drivers/i2csoft"
)

func main() {
	i2c := i2csoft.New()
	i2c.Configure(i2csoft.I2CConfig{
		Frequency: 400000,
		SDA:       machine.IO23,
		SCL:       machine.IO18,
	})

	sensor := adxl345.New(i2c)
	sensor.Configure()

	s := bme280.New(i2c)
	s.Configure()

	connected := s.Connected()
	if !connected {
		fmt.Println("BME280 not detected")
		return
	}

	for {
		x, y, z, _ := sensor.ReadAcceleration()
		fmt.Println("X:", x, "Y:", y, "Z:", z)

		rx, ry, rz := sensor.ReadRawAcceleration()
		fmt.Println("X (raw):", rx, "Y (raw):", ry, "Z (raw):", rz)

		temp, _ := s.ReadTemperature()
		println("Temperature:", strconv.FormatFloat(float64(temp)/1000, 'f', 2, 64), "C")

		time.Sleep(2000 * time.Millisecond)
	}
}
```

trevatk, 2021-09-23:
@sago35 and @aykevl 

Thank you both your help! Got everything working that I could with tinygo! Unfortunately I need bluetooth for the esp32 pico then I could use it for my sports tech project. 

Hope to be able to use this project in the future!

deadprogram, 2022-01-17:
Thanks for the feedback @trevatk now closing.

closed 2022-01-17 by deadprogram