tinygo-org/drivers #336: Lora sx126x and sx127x drivers

ofauchon, 2021-10-31:
Hi all. 

I've been very busy in the past month, but now I'd like to resume (and finish) my work on both sx126x and sx127x Lora drivers. 

As a reminder,  sx127x can be found on RFM95 Lora modules, and sx126x can be found in STM32WLE5 Lora SoC

Although there is still much work to get these drivers working, you can find here my current developpement branches:

https://github.com/ofauchon/tinygo-drivers/tree/lora-sx127x
https://github.com/ofauchon/tinygo-drivers/tree/lora-sx126x

If you are interested in helping in the developpement of these drivers, please leave a comment.

Tell us about your project, the hardware you plan to use (sx126x / sx127x), and how you can help (tests, developpement, providing hardware ? ).

Thanks

Olivier

ofauchon, 2021-11-02:
At first, the focus will be given to the completion of the sx127x driver. 

TODO : 

- Driver code cleanup and documentation
- Completion of the radio low level radio api.
- Extensive radio tests (eg: Power Levels / Boost/LDO/AGC/Frequency/SF/IQ...)
- Sample code for basic use of the driver (Lora packets RX / TX) 
- Support for both polling and interrupts when RX/TX

The next tests will be made with Bluepill + RFM95 modules, with the following wiring:

| Bluepill | RFM95 |
| ------- | ------- |
| GND | GND |
| Vcc | 3.3v |
| PA9 | RST |
| PA2 | DIO0 |
| PA5 | SPI SCK |
| PA6 | SPI MISO |
| PA7 | SPI MOSI |
| PA10 | SPI NSS |

Any comments, ideas welcome.

Olivier

JAicewizard, 2021-11-03:
I'm making devices for buoys that measure wind speed, and maybe also water temperature, wind direction and other things to a base node. Then we can aggregate this data to see if its good sailing weather!

I just got my soldering fixed so I have 2 working setups now. One bluepill is fake so I cant flash with tinygo, but I can flash C++ and use it to test the real one.

ofauchon, 2021-11-03:
What is the problem with your bluepill clones and tinygo flashing?

Maybe we could fix that

ofauchon, 2021-11-03:
I soldered two boards, too : 

- Bluepill STM32F103
- SX1276 module (https://www.tindie.com/products/blkbox/sx1276-lora-module-for-arduino/)

[image]

[image]

I tried to run some code too, but there is a problème with SPI (no clean SPI clock signal).
Not sure it's related to tinygo or my setup.

I'll try to solve this problem tonight.

Olivier

JAicewizard, 2021-11-03:
I wish I had the luxery of a 2.5mm pin pitch sx1276 module :)

The flashing problem seems to actually have been somewhat fixed, it seems to be an OpenOCD issue. The idcode is diferent so I need to change the code in the OPenOCD config for it to work. I cant set it to 0 as suggested in some online forums, that causes issues flashing to both.

ofauchon, 2021-11-04:
Some news: 

I could make the SX127x driver work on my setup (at lease for sending Lora packets, not yet receiving)
The problem of SPI clock may be related to high speed clock configuration (default PLL 72Mhz).

I could fix that with disabling PLL and switching to 8Mhz HSE. 
Still investigating if the problem is related to the use of  Bluepill STM32F103 Clones or not. 

Olivier

JAicewizard, 2021-11-04:
I can communicate fine when using https://github.com/sandeepmistry/arduino-LoRa at 72 Mhz. So this sounds like a tinygo SPI issue.

ofauchon, 2021-11-04:
So you tested the communication with 2 bluepill + RFM95 (one sender - one receiver ? )
Can you confirm me your wiring during your tests ?

I'll try to fix the problem on tinygo in the next days.

JAicewizard, 2021-11-04:
I dont see anything connected to the VIN on your RFM95, I assume you connected that some other way?
I can send you pictures of my setups. It will be like those kids games where you have to find where the wire goes.

But since I am also not able to connect to the RFM95 using tinygo I assume we have the same problem.

ofauchon, 2021-11-04:
Hi,

You'll find  here: 
[blue_rfm95_tinygo_8Mhz.zip](https://github.com/tinygo-org/drivers/files/7479286/blue_rfm95_tinygo_8Mhz.zip)

... an archive with elf/bin/hex firmware for bluepill

After flashing, you should see LED blinking every 5 seconds (While sending a sample Lora TX packet).
The firmware sets BluePill clock to HSE 8Mhz (instead of HSE/PLL 72MHz) as a fix for SPI problems.

Please note I used the following wiring: 

```
	SPI_SCK_PIN = machine.PA5
	SPI_SDO_PIN = machine.PA7
	SPI_SDI_PIN = machine.PA6

	SPI_CS_PIN  = machine.PB8
	SPI_RST_PIN = machine.PB9
	DIO0_PIN    = machine.PA2

	UART_TX_PIN = machine.PA9
	UART_RX_PIN = machine.PA10
```

If you try to send Lora Packets with this firmware and receive with arduino, ensure you configure this way on arduino side : 

```
I/Q Inversion : false
Sync Word : 0x34 
Frequency:       868100000,
SpreadingFactor: 12,
Bandwidth:       125000,
CodingRate:      6,
TxPower:         17,
```

On the serial console, you should read: 

```
LoRa Sender Example
SPI baudrate  16
Module version 0x12
Setup Lora:
Freq:  868100000 CR:  6 SF:  12
Sending LoRa packets every 5 seconds...
Sending: TinyGo LoRa Sender: 0
...
Sending: TinyGo LoRa Sender: xxxx

```

Good Luck  for testing.

Olivier

ofauchon, 2021-11-05:
SPI problem was fixed by @JAicewizard  in TinyGO 
( See https://github.com/tinygo-org/tinygo/pull/2229 for details)

closed 2022-07-05 by deadprogram

deadprogram, 2022-11-17:
Hello @ofauchon did anything ever happen with the `sx127x` driver? :smile_cat:

axellse, 2025-07-12:
just wanted to add: i couldn't get the tx function to return properly with the sx1278 chip, and just saw a random "p" printed out over usb serial (im guessing its start of a panic). Not sure why this worked, but leaving the spi frequency unset (unlike the 500000 specified in the example) when configuring the spi bus fixed the problem

deadprogram, 2025-07-13:
@axellse is this using the drivers in the `release` branch? Also on which MCU are you using it?

axellse, 2025-07-13:
@deadprogram yes, the latest stable release (via scoop), with an rp2040 based board.