tinygo-org/drivers #348: Wasm Canvas Displayer

spearson78, 2021-12-08:
While working on a tinygo display driver I found it useful to be able to debug issues in the web browser using the wasm target,

I have created an implementation of the Displayer interface that draws to an HTML5 canvas. This makes it much easier to test changes in both tinydraw and tinyfont.

I also added a javascript function to retrieve mouse events to simulate a touch panel making it possible to test more interactive applications in the browser.
Is there any interest in adding such a driver to the tinygo drivers package?
I would understand if this is not included as it requires supporting javascript and html code.

aykevl, 2021-12-11:
We already support something like that, but in in a different way: https://play.tinygo.org/
The "PineTime (dev kit)" and "Phytec reel board" both have a display and can be used using the appropriate drivers for those chips (as if using them on the real device), that is, `tinygo.org/x/drivers/st7789` and `tinygo.org/x/drivers/waveshare-epd/epd2in13`. I've recently made the devices look more like the real devices and intend to also add support for these in the TinyGo extension to VS Code so that it's usable the IDE directly during development. But I have so many other things that I'd also like to do...

I'm not sure whether the drivers repo is the right place though? You could certainly write a package that does this and implements the `drivers.Displayer` interface independent of the drivers repo. (My personal feeling, others might have a different opinion).

spearson78, 2021-12-20:
I recently discovered the tinydisplay project (https://github.com/sago35/tinydisplay) I'm working to extend it with Touch support tp support my tesing use case. This is a much better solution for testing as it doesn't require the additional html and javascript code. Once I've finished my current project I'll package my HTML canvas code into a separate repo.