tinygo-org/drivers #353: mcp2515 SPI panic on arduino ATMEGA328P

jadefox10200, 2021-12-19:
I am getting the following error : 
```
panic: compiler: could not store type code number inside interface type code

goroutine 30 [running]:
github.com/tinygo-org/tinygo/transform.LowerReflect({0x9b1bc70})
	/Users/distiller/project/transform/reflect.go:185 +0x905
github.com/tinygo-org/tinygo/transform.Optimize({0xc00005cdc8}, 0xc00028a300, 0x2, 0x0, 0x5)
	/Users/distiller/project/transform/optimizer.go:91 +0x4a5
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x796f980}, 0xc00028a300)
	/Users/distiller/project/builder/build.go:740 +0x1ed
github.com/tinygo-org/tinygo/builder.Build.func2(0xc0014c2f00)
	/Users/distiller/project/builder/build.go:386 +0x657
github.com/tinygo-org/tinygo/builder.jobWorker(0x246, 0x0)
	/Users/distiller/project/builder/jobs.go:182 +0xad
created by github.com/tinygo-org/tinygo/builder.runJobs
	/Users/distiller/project/builder/jobs.go:101 +0x1a5
```

My code below works on a pico. The panic is happening when I try to flash it: 

```
package main

import (
	"machine"
	"time"

	mcp "tinygo.org/x/drivers/mcp2515"
)

func main() {

	cs10 := machine.D10

	err := machine.SPI0.Configure(machine.SPIConfig{
		Frequency: 115200,
	})

	if err != nil {
		println(err)
		return
	}

	// //pointer to a Device is returned. Must pass chip select pin and the SPI0
	can := mcp.New(machine.SPI0, cs10)
	//
	// //configure the mcp2515 device
	can.Configure()
	//
	// //start it:
	//(d *Device) Begin(speed byte, clock byte) error {
	err = can.Begin(mcp.CAN500kBps, mcp.Clock8MHz)
	if err != nil {
		println("Error on begin")
		println(err)
	}
	//
	// //loop through and check for messages:
	for {
		if can.Received() {
			msg, err := can.Rx()
			if err != nil {
				println("ERROR ON RCV:")
				println(err)
				continue
			}
			println("GOT DATA:")
			println(string(msg.Data))
			time.Sleep(500 * time.Millisecond)
		}
	}

}

```

aykevl, 2021-12-28:
This is a limitation of the current reflect design. I hope to rewrite it eventually which would make this issue go away.
Meanwhile, the error that's displayed is not very user friendly and could certainly be improved.

aykevl, 2021-12-28:
Duplicate of https://github.com/tinygo-org/tinygo/issues/2414.

closed 2021-12-28 by aykevl