tinygo-org/drivers #362: SH1106 OLED Driver

jeffmhastings, 2022-01-17:
I didn't see a driver for the SH1106 OLED display driver/controller.

I'm interested in working with an [Adafruit Macropad RP2040 ](https://www.adafruit.com/product/5100) in tinygo. This board has an SH1106 OLED display connected to one of the SPI busses on the rp2040.

I'm probably going to start playing around with this, and I found some documentation for tinygo drivers which seems like a good starting point. Are there any other similar device drivers that might be a good reference for how to implement the driver for the SH1106?

sago35, 2022-01-19:
@jeffmhastings 
For example, the following may be helpful.

* SPI
    * https://github.com/tinygo-org/drivers/tree/release/ssd1351
    * https://github.com/tinygo-org/drivers/tree/release/examples/ssd1351
* SPI + I2C
    * https://github.com/tinygo-org/drivers/tree/release/ssd1306
    * https://github.com/tinygo-org/drivers/tree/release/examples/ssd1306

jeffmhastings, 2022-02-02:
Thanks for the examples. I've got something working, and I'd like to clean it up a bit before submitting a PR. I also have a few questions about some design considerations.

Is there an interface or other API specification that an OLED driver should conform to? It looks like both of those have a SetPixel function, and it looks like this is currently the only display function used in [tinydraw](http://github.com/tinygo-org/tinydraw). Is that the only function that should be implemented for an initial contribution? I also see ClearBuffer, ClearDisplay, Display, FillScreen, FillRectangle, etc. I'm just looking for some guidance on what sort of API should be implemented for this driver.

The other difference I noticed between the two examples is that the ssd1306 uses a buffer, but the ssd1351 doesn't. What are the considerations when deciding whether or not to use a buffer for the display data?

sago35, 2022-02-02:
> Edit: I found the Displayer interface. I'll implement this.

👍 

> The other difference I noticed between the two examples is that the ssd1306 uses a buffer, but the ssd1351 doesn't. What are > the considerations when deciding whether or not to use a buffer for the display data?

I don't know the right reason.
However, I think the following is an important point.

* Whether an arbitrary 1 pixel can be rewritten by SetPixel().
* Size required for buffer

sago35, 2022-02-16:
#376

jeffmhastings, 2023-01-29:
Closing as there's now an SH1106 driver introduced in #376.

closed 2023-01-29 by jeffmhastings