tinygo-org/drivers #363: image/jpeg and image/png tests fail to build?

dkegel-fastly, 2022-01-20:
I just cloned the repo and tried
```
cd image/jpeg
tinygo test
```

This failed with 
```
reader_test.go:92:2: wrong number of return values (want 2, got 1)
```

It looks like an internal disagreement about the return values of Decode()... @sago35, could you comment?

sago35, 2022-01-20:
@dkegel-fastly 
I changed the interface from Go's image package.

https://github.com/tinygo-org/drivers/tree/release/examples/ili9341/slideshow
https://github.com/tinygo-org/drivers/blob/90a13a697bc51b41b878ffc9b0fb5ff201ef0731/examples/ili9341/slideshow/main.go#L66-L76

However, it failed because I did not change the tests.

I think we can either maintain the test or remove it.

dkegel-fastly, 2022-01-21:
I'd vote for fixing the test and turning it on in CI, on the theory that it helps prevent bitrot.

deadprogram, 2022-01-22:
> I changed the interface from Go's image package.

Is that strictly needed? I have not looked at the specifics in this case, just asking.

sago35, 2022-01-24:
> Is that strictly needed? I have not looked at the specifics in this case, just asking.

Due to memory size limitations, it was not possible to return `image.Image`.
I changed the function signature to make the user aware of the mistake, but in hindsight I think it would have been better to just return nil.

* Go's image/png : `func Decode(r io.Reader) (image.Image, error)`
* tinygo.org/x/image/png : `func Decode(r io.Reader) error`

> I'd vote for fixing the test and turning it on in CI, on the theory that it helps prevent bitrot.

I'll take a look.