tinygo-org/drivers #373: Import paths are outdated?

jmacwhyte, 2022-02-13:
All throughout the project the import path "tinygo.org/x/drivers" is being used, but this URL now gives a 404. I'm assuming they all should be changed to "github.com/tinygo-org/drivers," but before I submit a PR to do that I want to make sure I'm not missing something, as it seems like a pretty basic thing that should have already been fixed.

Is my assessment correct that these links are broken and should be updated?

conejoninja, 2022-02-14:
Hello, you should "go get" the whole drivers repo, not an individual driver, this should work 

```bash
go get tinygo.org/x/drivers
```

jmacwhyte, 2022-02-14:
@conejoninja You are right! Sorry about that, just a stupid mistake on my part while trying to troubleshoot something else.

closed 2022-02-14 by jmacwhyte