tinygo-org/drivers #375: Support PWM for passive buzzers?

neildavis, 2022-02-14:
The 'buzzer' package generates the waveform in code using simple high/low switching. Many devices have hardware PWM capabilities exposed through the `machine` package. e.g. `PWM0`-`PWM7` are available from `machine_rp2040_pwm.go`
Would it be more efficient to make use of this?
I'm new to TinyGo and don't have experience with many different chips/boards (only Arduino Uno R3 & Pi Pico) so I'm not sure how supported it is and if it's worthwhile including?

I've written my own [PWM driver](https://github.com/neildavis/pi_pico_tinygo_examples/tree/main/modules/passivebuzzer) for use on the Pico which could form the basis of a PR if desired?

sago35, 2022-02-16:
@neildavis 
That's a very good code.

I created a PR #372 a while ago.
If you have any feedback, I'd love to hear it.

neildavis, 2022-02-23:
@sago35 
Thanks for your attrention. It looks like we want to achive the same thing :)

I have no objection to reviewing your PR #372, although I am fairly new to TinyGo, embedded, MPUs and Go in general (I have used Go a little in web services), so might not be the best reviewer. 

In general, what is the procedure for getting PRs reviewed & merged here? Do you know? I can't see from github who are the maintainers of this repo? Are we expected to request reviews from certain people? There's nothing in the `CONTRIBUTING.md` to suggest that?

neildavis, 2023-01-17:
Closing since 'tone' driver does this

closed 2023-01-17 by neildavis