tinygo-org/drivers #378: ./examples/wifinina/ntpclient cannot run

sago35, 2022-02-17:
The following message appeared and an error occurred.
Can anyone who can try it please run an example?

```
wifinina error: 0xf1
```

```
$ tinygo flash --target pyportal --size short --tags wifinina ./examples/wifinina/ntpclient/
```

This error will be output in the following locations

https://github.com/tinygo-org/drivers/blob/a15f2167cc7e46864d82e08904a3ddaa3e610d5e/wifinina/wifinina.go#L529-L539

doniacld, 2022-07-15:
I am encountering the same problem and had to overload the IP address. 

## Steps to reproduce the problem

### 1. Launch a web server
Launched a web server with the IP address of my computer with the following code:
```go
package main

import (
	"encoding/json"
	"fmt"
	"net/http"
)

//  curl -X POST -H "Content-Type: application/json" [http://localhost:80/measure](http://localhost/measure)  -d '{"temp": 270, "hum": 300}'
//  curl -X GET -H "Content-Type: application/json" [http://localhost:80/hi](http://localhost/hi)

type PostMeasureResponse struct {
	Temp int `json:"temp"`
	Hum  int `json:"hum"`
}

func decodeBody(w http.ResponseWriter, r *http.Request) {
	var m PostMeasureResponse

	err := json.NewDecoder(r.Body).Decode(&m)
	if err != nil {
		http.Error(w, err.Error(), http.StatusBadRequest)
		return
	}

	fmt.Printf("Temperature: %02d.%d°C, ", m.Temp/10, m.Temp%10)
	fmt.Printf("Humidity: %02d.%d%%\n", m.Hum/10, m.Hum%10)
}

func main() {
	fmt.Println("Listening on port 80")
	mux := http.NewServeMux()

	// POST /measure
	mux.HandleFunc("/measure", decodeBody)

	// GET /hi
	mux.HandleFunc("/hi", func(w http.ResponseWriter, r *http.Request) {
		fmt.Fprintf(w, "Hello, you")
	})

	err := http.ListenAndServe(":80", mux)
	if err != nil {
		panic(err)
	}
}
```

### 2. Send HTTP request from the board to the webserver
```go
// This example opens a TCP connection using a device with WiFiNINA firmware
// and sends a HTTP request to retrieve a webpage, based on the following
// Arduino example:
//
// https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/WiFiWebClientRepeating/
//
// This example will not work with samd21 or other systems with less than 32KB
// of RAM.  Use the following if you want to run wifinina on samd21, etc.
//
// examples/wifinina/webclient
// examples/wifinina/tlsclient
//
package main

import (
	"bufio"
	"fmt"
	"machine"
	"strings"
	"time"

	"tinygo.org/x/drivers/net"
	"tinygo.org/x/drivers/net/http"
	"tinygo.org/x/drivers/wifinina"
)

// access point info
const (
	ssid = ""
	pass = ""
)

// IP address of the server aka "hub". Replace with your own info.
// Can specify a URL starting with http or https
// const url = "http://tinygo.org/"
const url = "[http://192.168.0.20:80/hi](http://192.168.0.20/hi)"

// these are the default pins for the Arduino Nano33 IoT.
// change these to connect to a different UART or pins for the ESP8266/ESP32
var (

	// these are the default pins for the Arduino Nano33 IoT.
	spi = machine.NINA_SPI

	// this is the ESP chip that has the WIFININA firmware flashed on it
	adaptor *wifinina.Device
)

var buf [0x400]byte

var lastRequestTime time.Time
var conn net.Conn

func setup() {
	// Configure SPI for 8Mhz, Mode 0, MSB First
	spi.Configure(machine.SPIConfig{
		Frequency: 8 * 1e6,
		SDO:       machine.NINA_SDO,
		SDI:       machine.NINA_SDI,
		SCK:       machine.NINA_SCK,
	})

	adaptor = wifinina.New(spi,
		machine.NINA_CS,
		machine.NINA_ACK,
		machine.NINA_GPIO0,
		machine.NINA_RESETN)
	adaptor.Configure()
}

func main() {

	setup()
	http.SetBuf(buf[:])

	waitSerial()

	connectToAP()

	// You can send and receive cookies in the following way
	// 	import "tinygo.org/x/drivers/net/http/cookiejar"
	// 	jar, err := cookiejar.New(nil)
	// 	if err != nil {
	// 		return err
	// 	}
	// 	client := &http.Client{Jar: jar}
	// 	http.DefaultClient = client

	cnt := 0
	for {
		// Various examples are as follows
		//
		// -- Get
		// 	resp, err := http.Get(url)
		//
		// -- Post
		// 	body := `cnt=12`
		// 	resp, err = http.Post(url, "application/x-www-form-urlencoded", strings.NewReader(body))
		//
		// -- Post with JSON
		// 	body := `{"msg": "hello"}`
		// 	resp, err := http.Post(url, "application/json", strings.NewReader(body))

		resp, err := http.Get(url)
		if err != nil {
			fmt.Printf("%s\r\n", err.Error())
			continue
		}

		fmt.Printf("%s %s\r\n", resp.Proto, resp.Status)
		for k, v := range resp.Header {
			fmt.Printf("%s: %s\r\n", k, strings.Join(v, " "))
		}
		fmt.Printf("\r\n")

		scanner := bufio.NewScanner(resp.Body)
		for scanner.Scan() {
			fmt.Printf("%s\r\n", scanner.Text())
		}
		resp.Body.Close()

		cnt++
		fmt.Printf("-------- %d --------\r\n", cnt)
		time.Sleep(10 * time.Second)
	}

}

// Wait for user to open serial console
func waitSerial() {
	for !machine.Serial.DTR() {
		time.Sleep(100 * time.Millisecond)
	}
}

// connect to access point
func connectToAP() {
	time.Sleep(2 * time.Second)
	println("Connecting to " + ssid)
	err := adaptor.ConnectToAccessPoint(ssid, pass, 100*time.Second)
	if err != nil { // error connecting to AP
		for {
			println(err)
			time.Sleep(1 * time.Second)
		}
	}

	println("Connected.")

	ip, _, _, err := adaptor.GetIP()
	for ; err != nil; ip, _, _, err = adaptor.GetIP() {
		message(err.Error())
		time.Sleep(1 * time.Second)
	}
	message(ip.String())
}

func message(msg string) {
	println(msg, "\r")
}
```

### Error observed
```bash
~/workspace/src/github.com/doniacld/tinygo-discovery main* ❯ go run utils/read_serial.go -port /dev/cu.usbmodem11101                                                                                                                                            16:41:50
Let's begin
Connecting to myfree
Connected.
192.168.0.21
Connecting to MQTT...
We have a new client
wifinina error: 0x01
wifinina error: 0x01
wifinina error: 0x01
wifinina error: 0x01
```

Conversation slack when @ysoldak found the problem and a workaround for my case
https://gophers.slack.com/archives/CDJD3SUP6/p1656541040400669?thread_ts=1656344299.313069&cid=CDJD3SUP6

damongolding, 2022-12-08:
Hey @doniacld,

How did you get around the IP not being found issues. I am getting the same issue 😭

deadprogram, 2024-12-17:
Closing since this was all refactored. Thanks!

closed 2024-12-17 by deadprogram