tinygo-org/drivers #380: HD44780 WriteOnly mode not working

neildavis, 2022-02-19:
Has anybody successfuly used the `hd44780` driver in 'write only' mode? i.e. permanently holding the RW pin at GND and passing `machine.NoPin` as `rw` when [creating the `Device`](https://github.com/tinygo-org/drivers/blob/release/hd44780/hd44780.go#L66)

I wanted to use it this way since the [LCD board](https://www.displayfuture.com/Display/datasheet/alphanumeric/LCM1602A.pdf) I'm using requires 5V, but I'm interfacing with a 3.3V MCU (Pi Pico) so want to avoid reading from the device at 5V. 

I know I could use a bi-directional level shifter in the circuit, or a [hardware hack](https://www.codrey.com/electronic-circuits/hack-your-16x2-lcd/), but since the driver supported this mode and timing is not important to me I thought I'd try to use it.

I've tried in 8-bit and 4-bit modes but no luck. Just get the boxes instead of text. The [datasheet](https://www.displayfuture.com/Display/datasheet/alphanumeric/LCM1602A.pdf) for the LCD module says it uses a [SPLC780D](https://www.waveshare.com/datasheet/LCD_en_PDF/SPLC780D.pdf) rather than a  [HD44780](https://www.sparkfun.com/datasheets/LCD/HD44780.pdf) but from those datasheets I don't see a huge amount of difference in the timings.

I've tried changing `ClearHomeTime` & `InstrExecTime` in `Config` but found no values that work. 

I've also triued rebuilding the driver with various timing changes during the `Configure` function for the [initialization sequence](https://github.com/tinygo-org/drivers/blob/release/hd44780/hd44780.go#L116) to be more tolerant to the various devices as summarized [here](https://focuslcds.com/character-lcd-controller-compatibility-st7066/) but still no luck.

If I connect RW then it works flawlessly in both 8-bit and 4-bit mode, so I've ruled out hardware failure.

Any ideas greatfully recieved. Thanks.

Here's my program (8-bit mode variant)

```go
package main

import (
	"machine"
	"time"

	"tinygo.org/x/drivers/hd44780"
)

var (
	pinLCDEN   = machine.GP16
	pinLCDRS   = machine.GP17
	pinLCDData = [8]machine.Pin{machine.GP18, machine.GP19, machine.GP20, machine.GP21,
		machine.GP22, machine.GP26, machine.GP27, machine.GP28}
	pinLCDRW = machine.NoPin // Write only mode to avoid reading at 5V
	lcd      hd44780.Device
)

const (
	lcdCols          = 16
	lcdRows          = 2
	lcdClearHomeTime = time.Millisecond * 150
	lcdInstrExecTime = time.Microsecond * 150
)

func setupPins() {
	lcd, _ = hd44780.NewGPIO8Bit(pinLCDData[:], pinLCDEN, pinLCDRS, pinLCDRW)
	lcd.Configure(hd44780.Config{Width: lcdCols, Height: lcdRows,
		ClearHomeTime: lcdClearHomeTime, InstrExecTime: lcdInstrExecTime})
}

func main() {
	setupPins()
	lcd.SetCursor(0, 0)
	lcd.Write([]byte("Hello"))
	lcd.Display()
	lcd.SetCursor(0, 1)
	lcd.Write([]byte("World"))
	lcd.Display()
	for {
	}
}

```

deadprogram, 2022-04-21:
Was this addressed by https://github.com/tinygo-org/drivers/pull/361 ?

neildavis, 2022-04-25:
> Was this addressed by #361 ?

No, I only started using this repo at `v0.19.0` which is downstream of that commit (905fc6f). All the changes I tried from the descritpion were built against `dev` at the time.