tinygo-org/drivers #385: Add example with MCP3204

Gustavomurta, 2022-02-22:
I'm fascinated with Golang and TinyGo. I hope it will be the most used language in the very near future. I'm not an experienced programmer, not knowing how to use Github. I would like to contribute this example with the ADC converter MCP4204. Could someone include this example, please?

https://github.com/Gustavomurta/tinyGo_my_experiments/tree/main/MCP3204

sago35, 2022-02-22:
Hi @Gustavomurta 

You can make it the same way as below.
If you have any questions, we will be happy to help.

https://github.com/tinygo-org/drivers/pull/93
https://github.com/tinygo-org/drivers/blob/release/mcp3008/mcp3008.go

Gustavomurta, 2022-02-23:
Hi Sago,
I will try to do that as soon as possible.
Thank you