tinygo-org/drivers #386: Wifi issue: boards not consistent in whether reset is active high or low

slzatz, 2022-02-23:
When configuring the wifi on an Arduino Nano 33 IoT, the reset pin is active low; however, for the Arduino MKR 1010, the reset pin is active high.  (see [problem-with-nina...](https://forum.arduino.cc/t/problem-with-nina-w102-startup-on-mkr-1010/923644/14) for a somewhat confusing discussion.)  The result of this is that the Configure function in wifinina.go doesn't work for the MKR 1010.  Changing the function as below so it takes a parameter as to whether reset is active high or low does fix the problem -- not sure how you guys want to handle a change like this -- could create a second slightly renamed Configure function so it doesn't break anything using the current one.
```go
func (d *Device) Configure2(resetActiveHigh bool) {
   net.UseDriver(d)
   pinUseDevice(d)
   
   d.CS.Configure(machine.PinConfig{Mode: machine.PinOutput})
   d.ACK.Configure(machine.PinConfig{Mode: machine.PinInput})
   d.RESET.Configure(machine.PinConfig{Mode: machine.PinOutput})
   d.GPIO0.Configure(machine.PinConfig{Mode: machine.PinOutput})
   
   d.GPIO0.High()
   d.CS.High()
   d.RESET.Set(resetActiveHigh)
   time.Sleep(1 * time.Millisecond)
   d.RESET.Set(!resetActiveHigh)
   time.Sleep(1 * time.Millisecond)
     
   d.GPIO0.Low()
   d.GPIO0.Configure(machine.PinConfig{Mode: machine.PinInput})   
} 
```
If this makes sense to anyone happy to do a pull request.

Steve

racecarparts, 2022-06-20:
I have a MKR 1010, I added this to the end of `setup()` in [the connect example](https://github.com/tinygo-org/drivers/blob/dbff576c9ec1670c16e961624a43de936efdd045/examples/wifinina/connect/main.go#L27-L43) and got things working:

```go
adaptor.RESET.High()
time.Sleep(1 * time.Millisecond)
adaptor.RESET.Low()
```
result:

```go
func setup() {

	// Configure SPI for 8Mhz, Mode 0, MSB First
	spi.Configure(machine.SPIConfig{
		Frequency: 8 * 1e6,
		SDO:       machine.NINA_SDO,
		SDI:       machine.NINA_SDI,
		SCK:       machine.NINA_SCK,
	})

	adaptor = wifinina.New(spi,
		machine.NINA_CS,
		machine.NINA_ACK,
		machine.NINA_GPIO0,
		machine.NINA_RESETN)
	adaptor.Configure()

	adaptor.RESET.High()
	time.Sleep(1 * time.Millisecond)
	adaptor.RESET.Low()
}
```

onwsk8r, 2023-03-04:
Can this at least be added to some documentation somewhere? I've spent enough hours trying to figure out why WiFi didn't work that I could have written what I wanted to write in C by now. Probably in FORTRAN, too.

You have a big community of talented people who would love to help out - just tell us what to do.