tinygo-org/drivers #400: SPI driver for XPT2046

Arsen6331, 2022-04-12:
The XPT2046 touch controller supports SPI. The only driver in TinyGo is without SPI. Both implementations should exist (maybe add a function `NewSPI()` to use the SPI implementation). I would be happy to do this myself, but I have never written a driver for any component and haven't done very much embedded programming in general, so I don't know how. If someone can to point me in the right direction to do this or make it themselves, I would appreciate it.

gregoster, 2022-10-23:
Please have a look at: https://github.com/gregoster/tinygo-drivers/tree/xpt2046spi/xpt2046
There's still some extra design work needed to make both the GPIO and SPI versions co-exist in the same file (e.g. right now there it little error checking to make sure that SPI bits don't get called in GPIO mode and vice-versa).  
I built this driver to get a Waveshare Pico-ResTouch-LCD-2.8 work with both the st7789 and xpt2046 drivers at the same time, as t_clk, t_din, and t_dout end up being 'shared' between the GPIO and SPI, which didn't work for me... 
My repo also contains an SPI-only version of the driver (which is what I developed/used for testing).  I havn't tested the GPIO-only bits of this driver, so that would also need to be done.
This is my first venture into writing new hardware device drivers, and I must say I learned a fair bit (and had a lot of fun along the way).  Feedback/comments definitely welcome.

conejoninja, 2022-10-24:
Hi @gregoster nice work!
It would be great it you could make a PR from it, so we could properly review it and occasionally merge it.

A few general things I saw at first glance:
- Exported funcs  / vars need to be documented. Not required for un-exported, but highly recommended too.
- Please add an example for each (spi/gpio) in the examples/xpt2046 folder (you could re-use existing one)

gregoster, 2022-10-25:
I found a few cycles this evening to improve things.  Cleaned up the code a bunch, and hopefully have removed any 'sharing' issues between GPIO and SPI interfaces.  Also added a whole bunch of comments.  I also added an example using SPI: https://github.com/gregoster/tinygo-drivers/blob/xpt2046spi/examples/xpt2046/mainspi.go  If things are looking reasonable enough I'll next see about getting a PR created.  Thanks!

conejoninja, 2022-10-25:
Hello,

 Please, put each example inside a folder like
```
examples/xpt2046/gpio/main.go
examples/xpt2046/spi/main.go
```

Please, make a PR (to the `dev` branch), so we can comment and create a proper review of your work, don't worry if you need to make several fixes to your initial PR, that's completely normal.

gregoster, 2022-10-27:
Examples move to folders (should have though of that one myself, but I'm still a beginner when it comes to go and file layouts).  PR submitted: https://github.com/tinygo-org/drivers/pull/477  Thanks for helping walk me through things... lots to learn about this process.